guysoft/OctoPi #56: Octopi Local doesn't work, but device IP does

timo0060, 2015-01-11:
As the title says, I can access the print server interface by using the IP address of my raspberry pi, but not using http://octopi.local even if I use https://octopi.local. I don't want to have to remember the IP address every time I want to use this service. Thanks for the help.

guysoft, 2015-01-11:
What Browser?
What Operating system?
Also what router is being used in your network and is it running a custom firmware such as Open WRT?

timo0060, 2015-01-12:
Sorry for the lack of information. I'm on windows 7 using Chrome v39, and also tried on Internet Explorer 11. As for the router, it's a modem/router combo. It's the 2701HG-G Gateway with Bell Canada's stock firmware on it. Anything I missed or anything else you need, let me know, because the place I want to set my printer up at isn't accessible to my computer.

dwatts418, 2015-01-15:
Try http://octopi.local:5000

timo0060, 2015-01-15:
That didn't do it either, unfortunately.

loadedsith, 2015-01-17:
Do you have Bonjour installed, via installing iTunes or Safari?

atifsatti786, 2015-01-19:
http://octopi.local never works. It is really frustrating. It all wasting of time and money. I will not recommend anyone to buy raspberry pi for 3d printer.

guysoft, 2015-01-20:
@atifsatti786 IP works fine.
Reading on the net the 2701HG-G Gateway does not seem to come with one of those common stock firmwares big companies use with settings I can predict without ever seeing one. So I would guess it might not be resolving local hostnames normally. Some firmwares resolve https://octopi.lan and not https://octopi.local , try .lan first. However, they might be using something completely different all together. If you could log in and post some more information we could try and locate the setup.
This is not something I can solve on the OctoPi software side, I suggest you try pinging other hostnames on your networks, or ask people that have this router what they do. And share here so we know for next time :)

myoung34, 2015-01-21:
This to me sounds like a DNS issue to the router.
can you try via bash $ dig @routerip octopi.local
Your best bet is to see if you can add an A record in your router for octopi.local to the IP of your pi in the DNS settings for your router. I do this via an active directory server and point my routers primary DNS to this server.

timo0060, 2015-01-28:
Sorry for the late reply. I've been quite busy over the past few days. octopi.lan doesn't work either unfortunately. I also pinged other host names on my network, such as my laptop. My desktop will just time out when trying to ping any hostname, but my laptop keeps getting responses from the same ip address no matter which host I ping. Is this because my laptop is wireless and my desktop isn't? 

Also, how do I "bash $ dig @routerip octopi.local"?

guysoft, 2015-02-09:
@timo0060 Its a small program that comes with linux. You can also install it in windows [googled instructions here](http://www.danesparza.net/2011/05/using-the-dig-dns-tool-on-windows-7/)

seefood, 2015-03-15:
Also, if the router is too dumb to do bonjour and other tricks, it may still be possible to lock the Raspi's IP (per its MAC) and give it a name in the /etc/hosts (yes, Windows has one too, it's just a bit hidden, can't remember where)

Either way, @guysoft is correct, that's not something the Raspberry can have control over, it's the router that is not smart enough, most probably...

myoung34, 2015-03-17:
this should def be closed, it's not to do with octopi. @seefood for reference: ``C:\windows\system32\drivers\etc\hosts``

closed 2015-07-05 by foosel

techknight, 2016-01-15:
The simplest fix for this IMO is to install Samba on your Pi by doing this:
```
sudo apt-get install samba
```

You don't have to add any shares or configure anything, but the Windows devices on your network will then be able to ping it or connect to http://octopi in your web browser.

If you have trouble seeing it on Windows after installing Samba, you may have to do this first from an administrative command prompt on your Windows device:
```
ipconfig /flushdns
```

myoung34, 2016-01-15:
@techknight the samba package is only part of that. It includes `nmbd` which is the server that responds windows resolution. If he's on OSX/Linux it wont make a difference since ping uses IGMP.

The easiest solution is to put it in hosts or use DNS on his router (most support this)

techknight, 2016-01-16:
@myoung34 Oh! maybe just installing the nmdb package would be better, then - I just want Windows folks to know that instead of configuring their router (which may have a variety of feature levels and UIs), or editing the hosts file,  there's another option that will probably meet their needs with one command on the Pi.

chriskimbr, 2019-02-10:
> The simplest fix for this IMO is to install Samba on your Pi by doing this:
>
> ```
> sudo apt-get install samba
> ```
> You don't have to add any shares or configure anything, but the Windows devices on [...]

That worked! TY

John-Mc, 2019-02-10:
I set my OctoPi up to email me it's IP address each time it boots up. Normally, I access is via a Mac, which comes with Bonjour installed, so using octopi.local works with no further hassle. However, when I was helping out at the local elementary school, there was no telling which ancient laptop we were going to use to access the Pi, and the school had them locked down, so we could not install Bounjour/ZeroConf on them. IF we had a problem, I'd just check my email for the IP address.

rakeshbarman, 2019-06-10:
I have been using octopi.home . Try it out.

SeventhDread, 2019-06-10:
Hi I just use the IP address but if you want to use octoprint.local you have to install and set up bonjour from Apple as no windows browsers can use .local and instead of using internet explorer I use Microsoft edge as some internet explorer stuff is blocked like you can’t view camera on internet explorer but if you really want to use .local use bonjour but your computer should remember the IP address over time.

samuelkw1999, 2026-08-19:
Had the same issue and realized what was happening and fixed it

when flashing the image to the pi in raspberry pi imager I set the device name to something other than octopi.
So octopi local wasn't connecting because its no longer 'octopi.local' but rather 'devicename.local'
for instance if i named it squidpi then it isn't supposed to connect to octopi.local, its supposed to be squidpi.local instead

you can see what yours is by plugging your octopi into a monitor and logging in using the username and password you set up in raspberry pi imager. it should show 

"Access Octoprint from a web browser on your network by navigating to any of 
http://devicename.local
http://ipaddresshere

https is also available, with a self-assigned certificated"

hope this helps someone