guysoft/OctoPi #57: Loses keyboard and mouse after configuration.

bshirley, 2015-01-14:
I took the image from the mirror site. Installed it on a micro SD. Booted my RaspberryPi B+, with wifi dongle, and mac keyboard/mouse.  I did this with the image that came with the RaspberryPi with no problem. When I do it with the OctoPi image, after the minimal configuration (expand to use whole SD card, and auto boot into UI - first time it was to command line) the keyboard and mouse are not recognized after the reboot.  Without that, can't attach to wifi, can't get further. 

2014-09-09 0.10.0

guysoft, 2015-01-14:
@bshirley Just to make sure - do you mean the keyboard and mouse are used on a mac, in the browser?
Are you in the devel branch or master?

Also cc @foosel who just posted about they keyboard feature on g+

foosel, 2015-01-14:
@guysoft unrelated, first of all what I posted on G+ is only relevant to the devel branch for now (and the above doesn't sound like the OP upgraded to that) and also is a purely client related part that can't influence mouse and keyboard recognition at all.

bshirley, 2015-01-14:
The keyboard/mouse were plugged into the RPi.

I got it to work the third time. I changed the configuration items I played with and also unchained the keyboard/mouse.  I plugged them each into their own USB plug.  Not sure that the final fix actually was (changing too many variables at once), and it takes so long to push the image onto the microSD that I'm not going to experiment.

Thanks for the reply.

(I'm now having an issue with the OctoPi communicating with my Type A Series 1 2012 - but I need to spend more time debugging that, possibly hardware/firmware related.)

guysoft, 2015-01-17:
I am not sure I understand how you can control the printer with a keyboard that is connected to the Pi?
I mean OctoPrint runs on a browser, input from the keyboard should come from the computer the browser is running on. Are you opening a browser on the Raspberrypi too?

foosel, 2015-01-17:
I think he just meant that keyboard and mouse were not working for him on
the pi, like when connecting it to a monitor and trying to use either it
wouldn't work, eg on the command line or X. Don't see anything in the
original ticket about printer controls.

bshirley, 2015-01-20:
correct, foosel. 

I was running the pi with a monitor plugged into the HDMI, a keyboard/mouse via USB - no interfacing with the printer yet.  that problem fixed itself when i modified one or more of several variables.

(my current problem involves interfacing with the printer, i haven't had time to debug enough to provide useful information)

foosel, 2015-01-20:
Just for the record then: [[FAQ] I can't get OctoPrint to connect to my printer](https://github.com/foosel/OctoPrint/wiki/FAQ#i-cant-get-octoprint-to-connect-to-my-printer)

foosel, 2015-07-05:
Closing since fixed for the OP, guess it was some odd power problem or something like that.

closed 2015-07-05 by foosel