guysoft/OctoPi #58: [Enhancement] provide means to control camera's enablement via a lock file

wolfspyre, 2015-01-23:
Hi there,

I have a proposed means to provide a way to both improve the efficiency of the camera script, but also provide a lock file which can disable/enable the camera that I'd like to run by you

if you look at scripts/webcamDaemon, in your while loop you always fire the start functions if the camera is present. 
if we additionally validate that the process isn't already running before running the start(Usb|Raspi) functions we could save a bunch of IO and executions. Granted, it's only once every 120s, but still.

additionally, if you wrap both of the enablement if stanza's in another if statement which checks for the existence of the file /home/pi/.octoprint/cam_enabled,
you could decide at that point if you should start, or stop the camera process. This would give you a way to control this process fairly easily with the api/gui by altering the existence, or lack thereof of /home/pi/.octopi/cam_enabled

Thoughts? I'm happy to do the script work and sub a PR if you like the idea.

guysoft, 2015-01-23:
Sounds like a good feature. A pull request would be welcome.
It might be nice to add an OctoPrint plugin for this option cc @foosel
Note I am busy so I can'y write it. Will review it.

wolfspyre, 2015-02-05:
ok, here's the idea I've got for the while loop logic:

while true; do
  #figure out our variables

  #check to see if process should be running
  #when no
   # is process running?
     #no
       #we're done
     # yes
       # check to see if the kill_it_with_fire data exists
         # YES! do the needful! Kill it with fire
         # Nope! we think the universe is sane
       # is a timelapse in progress?
         # yes. wait til timelapse is done.
           # configurable?
         # timelapse not in progress
           #stop camera process here
  #when yes
    # is the process running?
      #yes!
        # yes exactly as we need it to
          # Great! We're done. remove kill it with fire data
        # yes it's running, but not how we need it to
          # Crap. Kill it with fire. die, let next invocation of kill loop handle this
      #not running. needs be
        # start it
        # remove kill it with fire data
sleep 120
done

I've got most of this pretty easily. The one bit I'm uncertain of is 'how do I tell if a time-lapse is in progress?'

I've a multifaceted reason for this question; I'm having problems getting timelapses to actually work on my octopi rig, (they did originally, and somehow I broke it); and figured I'd kill two birds with one stone and implement logic in the script so that if we change the options in the gui, we can restart the webcam daemon automagically at a time that it's not going to interrupt a recording-in-progress.

guysoft, 2015-02-05:
There are too many ifs there, how does that help anyone? (For the record I have 0 time to write this even if it does).
Even this psudocode is hard to understand, maintaining such a thing is even worse.

wolfspyre, 2015-02-06:
Sorry, I was trying to explain the pseudo code.

I'll write the script and explain the logic more succinctly 

MobileMail