guysoft/OctoPi #60: RPi 2 support

MoonshineSG, 2015-02-06:
will there be a build for the new RPi 2 ?

guysoft, 2015-02-06:
Hey,
There are nightly builds available at http://docstech.net/OctoPiMirror/nightly/
They should work with RPi 2. I do't have one to test, and it would be great if someone will confirm he it works, and which nightly build

emericklaw, 2015-02-06:
I have used the 06-02 nightly on a new RPi 2 but get 405: Method Not Allowed when requesting the main page. I tried to access other resources like images and style sheets and all I tried failed.

Networking and SSH have worked fine, yet to test the webcam.

guysoft, 2015-02-06:
@foosel it might be the HAProxy rules that were changed?

guysoft, 2015-02-06:
I think #50 caused a regression. Not sure how to fix it :-/

guysoft, 2015-02-06:
@emericklaw Ok merged a fix from @foosel . It should build in this nightly, and you can test it out. Or just apply it yourself and test.

emericklaw, 2015-02-07:
I updated the haproxy.cfg and after a reboot all appears to be fine, I have not been able to test connecting to a printer yet.

guysoft, 2015-02-07:
2015-01-31-octopi-0.11.0-2015-02-07.zip build should have the fix

closed 2015-02-09 by guysoft