guysoft/OctoPi #62: Installing OctoPrint Plugins fail

eboston, 2015-02-07:
I was trying to install some OctoPrint plugins and the install fails because it cannot find required items (like OctoPrint) installed.  I'm guessing it might have something to do with the way OctoPi is created (Python virtual work areas - or something like that) vs a straight OctoPrint install.  Is there a special trick that needs to be done to get the plugins to install?

Running Version: 1.2.0-dev-444-g922ee39 (devel branch) OctoPrint.  
OctoPi installed from 2014-09-09-wheezy-octopi-0.10.0.zip

guysoft, 2015-02-09:
Can you write here what commands you are using in the terminal and what error are you getting?
Just a wild guess since I don't know what you are doing, you might need to call python binary at:
`~/oprint/bin/python`

eboston, 2015-02-09:
Plugin was OctoPrint-Atcommands (https://github.com/dattas/OctoPrint-Atcommands).  I used the instructions and issued the command: 

pip install https://github.com/dattas/OctoPrint-Atcommands/archive/master.zip

guysoft, 2015-02-09:
Try
``~/oprint/bin/pip install https://github.com/dattas/OctoPrint-Atcommands/archive/master.zip``

Does that work?

eboston, 2015-02-09:
I will give that a try but might be a day or two before I can get to it.

eboston, 2015-02-10:
Got my new Raspberry Pi 2 today.  Installed the latest nightly, switched to devel branch of OctoPrint and the command you gave worked.

closed 2015-02-11 by guysoft