guysoft/OctoPi #63: Snakeoil

eboston, 2015-02-08:
I have been going through the different configuration files and have noticed references to 'snakeoil' in a few places.  One in particular is with the haproxy.cfg file.  On a system where OctoPrint has been built from scratch, the bind that references generates an error.  Is there any information available about that package and how it needs to be installed?

guysoft, 2015-02-08:
Yes,
That package is part of debian packages called ``make-ssl-cert`` and ``generate-default-snakeoil`` . [More information available here](http://askubuntu.com/questions/49196/how-do-i-create-a-self-signed-ssl-certificate).
Its a package that generated a self-signed certificate for OctoPi.
When you say you are on a "systerm where OctoPrint has been built from scratch". Are you trying to build OctoPi? Or OctoPrint? Because if you are talking about OctoPrint and you found that entry in the wiki please point us to it.

eboston, 2015-02-08:
Was trying to build a Banana Pi version of OctoPi.  Was trying to get things working like they are on the Raspberry version and this was about the only thing I wasn't able to resolve.  I've haven't seen any issues without it installed other than the haproxy.cfg use but commenting that out fixes the problem.  Just trying to find out how critical it was to have.

But seeing that there is the new Raspberry Pi 2 Model B available, I'll probably just scrap the Banana Pi work.

guysoft, 2015-02-08:
Actually I would like to help you with that.
Did the mounting work? Because we do want to extend support to other machines including the Bannana Pi. I Just don't have one to test here.

What image are you using?

eboston, 2015-02-08:
Is there a way to send private messages on here?  I have something to post that I don't want out in public.

guysoft, 2015-02-08:
@eboston  guysoft [at] gmail.com :)
Also works with hangouts and you can look me up

eboston, 2015-02-10:
Not sure if my emails have been making it through to you or not, but have sent you two requesting the address you want the Banana sent to.  If you are still interested in it, send me the address to ship it to and I'll get it right out.

guysoft, 2015-03-04:
@eboston Were you trying to use bananapi raspbian? Lubuntu or Bananian to build?

eboston, 2015-03-04:
The Bananapi raspbian. I was able to get everything installed and running
with OctoPrint. Only thing I didn't was your webcam daemon and the snakeoil
stuff.

guysoft, 2015-03-07:
Hey,
Ran a nightly build, snakeoil works fine on their last raspbian image. The camera indeed does not work. I am getting:
    
    pi@octopibanana ~/mjpg-streamer $ sudo modprove ov5640
    sudo: modprove: command not found

Not sure how to fix this yet.

I wrote a bananapi variant using the new variant build system, so hopefully we could get bananapi builds for octopi pretty soon.

nophead, 2015-03-07:
modprove should be modprobe I guess?

guysoft, 2015-03-08:
Ops, pasted the wrong line, fixed

eboston, 2015-03-08:
So got it all working on the banana pi now?

guysoft, 2015-03-08:
No, I get:

    FATAL: Module ov5640 not found

I fixed the right error message from the command line :-/
Are there any instructions how to get the Bpi-camera working on the raspbian/bananian distro?

eboston, 2015-03-08:
These are the instructions I followed and was able to get the camera to work - after I plugged it into the correct connector.  The connectors are in the opposite position on the Banana than they are on the Raspberry.

http://wiki.lemaker.org/Camera_Module

guysoft, 2015-03-08:
@eboston 
1. The section "Setting up the B-Pi" starts with that command, that fails. 
2. the source code link to the module is a broken link

eboston, 2015-03-09:
Did you start with the image here: http://www.lemaker.org/resources/9-81/raspbian_for_bananapi.html?  Be sure to get the non-pro version.  I don't remember having to install the module.  It was already there.

guysoft, 2015-03-09:
No, I used the image here:
http://bananapi.com/index.php/download?layout=edit&id=29

Because its from 08/02/2015 which seemed the most up to date.

Could you explain why I should take the older version?

eboston, 2015-03-09:
Since the one I pointed too has the module already, you might be able to extract it from that image and install it to the one you have.  The one I linked to is from the people that made the board.  I'm not sure of the relationship with the one you linked to, the one I linked to, and an official Banana Pi is.

guysoft, 2015-03-09:
@eboston You can't just extract the module because its not the same kernel version, and not the same kernel configuration.

eboston, 2015-03-09:
That might be a good reason then to use the one I linked to.  :)

guysoft, 2015-03-09:
I'll try using your image, but really it might cause new problems.

In the meantime here is a build from the Raspbian banana pi image people can try out:
http://docstech.net/OctoPiMirror/nightly/2015-01-31-octopi-bpi-r1-0.12.0-2015-03-09.zip

Is there anyone in the BananaPi project we can coordinate with?

eboston, 2015-03-09:
I was able to get everything running but the snakeoil and web daemon script
using the image I linked to so hopefully using it won't create any
problems. As for coordinating with the Banana Pi folks, I have never talked
to them so maybe just email them with the issue about the module and see if
anyone responds.

guysoft, 2015-03-30:
Hey,
So to update I tried switching to the ``Raspbian_For_BananaPi_v1412.zip`` image.
It failed on setting up the ``ca-certificates-java`` package because `/proc` was not mounted.
Once I mount that I get now:

    Setting up ca-certificates-java (20121112+nmu2) ...
    Error occurred during initialization of VM
    Could not reserve enough space for 1048576KB object heap

If anyone knows how to solve this kind of error that would help

guysoft, 2015-04-11:
Ok, tried to extend the memory allocation on qemu-arm-static using:

    export QEMU_RESERVED_VA=0xf7000000
    chroot . usr/bin/qemu-arm-static /bin/bash /chroot_script

But now I get:

    Unable to reserve 0xf7000000 bytes of virtual address space for use as guest address space (check your virtual memory ulimit setting or reserve less using -R option)

There is no -R option in qemu-arm-static, or ulimit. Any ideas?

guysoft, 2015-04-12:
OK, found a solution to that, apparently the problem was not qemu, it was that this image has a bad build of Java. So this solves it:

    mount -t proc none /proc
    apt-get install -y openjdk-7-jre
    update-java-alternatives  -s java-1.7.0-openjdk-armhf
    apt-get install -y ca-certificates-java
    umount /proc

@eboston How did you get everything else working if the username is bananapi and not pi? In the Lemaker image all hard-coded in the scripts. Did you manually install OctoPrint, or did you run the build scripts?

eboston, 2015-04-12:
I manually installed everything.

guysoft, 2015-04-27:
Switched back to 08/02/2015. Lemaker's image is extremely out of date and causes a lot of problems.
Its probably  better to go with the up-to-date image and rebuild the kernel module for the camera (which I am note sure how to do here).

At least the nightly builds are usable now.

guysoft, 2015-06-10:
Done, there are nightly builds for bananapi.

closed 2015-06-10 by guysoft