guysoft/OctoPi #65: Haproxy config

tiagojanz, 2015-02-09:
Hi,
I had to reinstall  my Octopi instalation, but because I had to update my system it seems that haproxy gone wild.

My haproxy version

pi@octopi ~ $ haproxy -vv
HA-Proxy version 1.5.8 2014/10/31
Copyright 2000-2014 Willy Tarreau <w@1wt.eu>

Build options :
  TARGET  = linux2628
  CPU     = generic
  CC      = gcc
  CFLAGS  = -O2 -g -fno-strict-aliasing
  OPTIONS = USE_LINUX_SPLICE=1 USE_LINUX_TPROXY=1 USE_OPENSSL=1 USE_PCRE=1

Default settings :
  maxconn = 2000, bufsize = 16384, maxrewrite = 8192, maxpollevents = 200

Encrypted password support via crypt(3): yes
Built without zlib support (USE_ZLIB not set)
Compression algorithms supported : identity
Built with OpenSSL version : OpenSSL 1.0.1e 11 Feb 2013
Running on OpenSSL version : OpenSSL 1.0.1e 11 Feb 2013
OpenSSL library supports TLS extensions : yes
OpenSSL library supports SNI : yes
OpenSSL library supports prefer-server-ciphers : yes
Built with PCRE version : 8.31 2012-07-06
PCRE library supports JIT : no (USE_PCRE_JIT not set)
Built with transparent proxy support using: IP_TRANSPARENT IPV6_TRANSPARENT IP_FREEBIND

Available polling systems :
      epoll : pref=300,  test result OK
       poll : pref=200,  test result OK
     select : pref=150,  test result OK
Total: 3 (3 usable), will use epoll.

when I try to run:

/usr/local/sbin/haproxy -f /etc/haproxy/haproxy.cfg
[ALERT] 039/122814 (2568) : parsing [/etc/haproxy/haproxy.cfg:31] : error detected while parsing a 'reqadd' condition : no such ACL : 'is-ssl'.
[ALERT] 039/122814 (2568) : Error(s) found in configuration file : /etc/haproxy/haproxy.cfg
[WARNING] 039/122814 (2568) : Setting tune.ssl.default-dh-param to 1024 by default, if your workload permits it you should set it to at least 2048. Please set a value >= 1024 to make this warning disappear.
[ALERT] 039/122814 (2568) : Fatal errors found in configuration.

My config file

  GNU nano 2.2.6                                                            File: /etc/haproxy/haproxy.cfg                                                                                                                                

global
        maxconn 4096
        user haproxy
        group haproxy
        daemon
        log 127.0.0.1 local0 debug

defaults
        log     global
        mode    http
        option  httplog
        option  dontlognull
        retries 3
        option redispatch
        option http-server-close
        option forwardfor
        maxconn 2000
        timeout connect 5s
        timeout client  15min
        timeout server  15min

frontend public
        bind *:80
        bind 0.0.0.0:443 ssl crt /etc/ssl/snakeoil.pem
        option forwardfor except 127.0.0.1
        use_backend webcam if { path_beg /webcam/ }
        default_backend octoprint

backend octoprint
        reqrep ^([^\ :]*)\ /(.*)     \1\ /\2
        reqadd X-Scheme:\ https  if is-ssl
        server octoprint1 127.0.0.1:5000

backend webcam
        reqrep ^([^\ :]*)\ /webcam/(.*)     \1\ /\2
        server webcam1  127.0.0.1:8080

guysoft, 2015-02-09:
Hey,
What version did you install?

guysoft, 2015-02-09:
@foosel The seems to show that #50 is still not working

foosel, 2015-02-09:
#50 is working just fine in the current nightly (just tested myself earlier today), but this haproxy.cfg up there is not from the current nightly. It's an older version from way before any fixing attempts.

tiagojanz, 2015-02-09:
Haproxy version?

pi@octopi ~ $ haproxy -vv
HA-Proxy version 1.5.8 2014/10/31
Copyright 2000-2014 Willy Tarreau w@1wt.eu

Octoprint is the latest

Heres the comand I run to update Devel

//UPDATE GIT
cd ~/OctoPrint
git pull && git checkout devel
~/oprint/bin/python setup.py clean
~/oprint/bin/python setup.py install
sudo service octoprint restart

foosel, 2015-02-09:
No, your OctoPi version this time

tiagojanz, 2015-02-09:
I installed this image   2014-09-09-wheezy-octopi-0.10.1.img
I then run 

sudo apt-get update
sudo apt-get upgrade

guysoft, 2015-02-09:
@tiagojanz That version was scrubbed because of HAProxy problems, please use:
http://docstech.net/OctoPiMirror/nightly/2015-01-31-octopi-0.11.0-2015-02-09.zip
Its currently the release candiate.
I am deleting 0.10.1, this is the second time its happens and it confuses people.

closed 2015-02-09 by guysoft