guysoft/OctoPi #66: USB webcam not working in nightly build

slipjig, 2015-02-10:
Hi,
I thought I'd report something I noticed recently.  I have a RPi B+ with an Edimax wifi adapter and a Microsoft Lifecam HD-5000, controlling a Printrbot Simple Metal, all connected to a powered USB hub with a 3.5A power supply.  I downloaded and installed the most recent release version Octopi image (0.10.0, from 09/09), which worked mostly great for a few days.  My SD card got corrupted at one point though, and I reinstalled using the nightly build from 1/31.

The 1/31 nightly build worked and I could print, but the USB webcam wouldn't work with mjpg-streamer.  I'm not sure what the issue was exactly...lsusb and dmesg showed the camera was recognized and connected without errors that I could see, but mjpg-streamer wasn't able to fetch images, so I'd get an X in place of the image in both its web pages as well as Octoprint's.

This evening I was in a hurry to get the webcam working so I could run a print tomorrow during work hours, so I reinstalled the release version of 0.10.0, and now the camera works again.  It seems something changed in the nightly builds that broke the webcam.  Maybe it's a config issue or something, but I think I set it up the same way both times.

In a few days I'll be in a position to reinstall the nightly build again if there are steps I can do to get more information, but I thought I'd post this in the interim.
-Mike

guysoft, 2015-02-11:
OK we have found what is causing it and it should be fixed in tonight's nightly build.

See comments at the end of this thread:
https://plus.google.com/u/0/+GuySheffer/posts/j3N1HiRiAcs

slipjig, 2015-02-11:
Thanks!  You guys rock 😊

Mike

Sent from Windows Mail

From: Guy Sheffer
Sent: ‎Wednesday‎, ‎February‎ ‎11‎, ‎2015 ‎9‎:‎57‎ ‎AM
To: guysoft/OctoPi
Cc: Mike Simpson

OK we have found what is causing it and it should be fixed in tonight's nightly build.

See comments at the end of this thread:
https://plus.google.com/u/0/+GuySheffer/posts/j3N1HiRiAcs

guysoft, 2015-02-13:
Hey, 
Can someone test the last two nighties and tell me if they work?
Last one is at: http://docstech.net/OctoPiMirror/nightly/2015-01-31-octopi-0.11.0-2015-02-13.zip

nophead, 2015-02-13:
I will try that one but it takes about 2 hours to download.

guysoft, 2015-02-13:
We are working on fixing that too ;)
For now we are living of donated bandwidth, so this is what we have.

nophead, 2015-02-13:
I don't have much download bandwidth so it probably takes that long anyway.

nophead, 2015-02-13:
It downloaded OK and boots on a RPI B2. The MD5 hash is 98e21c7cea5621de33deda8b6ae751e2.

A random USB webcam I plugged in does not work: -
```
Feb 13 18:28:17 octopi MJPG-streamer [2123]: starting application
Feb 13 18:28:17 octopi MJPG-streamer [2123]: MJPG Streamer Version: svn rev: Unversioned directory
Feb 13 18:28:17 octopi MJPG-streamer [2123]: Using V4L2 device.: /dev/video0
Feb 13 18:28:17 octopi MJPG-streamer [2123]: Desired Resolution: 640 x 480
Feb 13 18:28:17 octopi MJPG-streamer [2123]: Frames Per Second.: 10
Feb 13 18:28:17 octopi MJPG-streamer [2123]: Format............: JPEG
Feb 13 18:28:17 octopi MJPG-streamer [2123]: TV-Norm...........: DEFAULT
Feb 13 18:28:17 octopi MJPG-streamer [2123]: Frame period time ......: 100 ms
Feb 13 18:28:17 octopi MJPG-streamer [2123]: www-folder-path...: ./www/
Feb 13 18:28:17 octopi MJPG-streamer [2123]: HTTP TCP port.....: 8080
Feb 13 18:28:17 octopi MJPG-streamer [2123]: username:password.: disabled
Feb 13 18:28:17 octopi MJPG-streamer [2123]: commands..........: enabled
Feb 13 18:28:17 octopi MJPG-streamer [2123]: starting input plugin input_uvc.so
Feb 13 18:28:17 octopi MJPG-streamer [2123]: starting output plugin: output_http.so (ID: 00)
Feb 13 18:28:18 octopi kernel: [   14.796002] Transfer to device 4 endpoint 0x5 frame 1768 failed - FIQ timed out. Data may have been lost.
Feb 13 18:28:18 octopi kernel: [   14.861014] Transfer to device 4 endpoint 0x5 frame 1833 failed - FIQ timed out. Data may have been lost.
Feb 13 18:28:18 octopi kernel: [   15.058041] Transfer to device 4 endpoint 0x5 frame 2030 failed - FIQ timed out. Data may have been lost.
Feb 13 18:28:18 octopi kernel: [   15.122921] Transfer to device 4 endpoint 0x5 frame 47 failed - FIQ timed out. Data may have been lost.
Feb 13 18:28:18 octopi kernel: [   15.155038] Transfer to device 4 endpoint 0x5 frame 79 failed - FIQ timed out. Data may have been lost.

Lots of the same.
```

A Logictech USB webcam I have used before still works fine though.

guysoft, 2015-02-13:
So it works but spits errors?

nophead, 2015-02-13:
Logitech works fine without errors. Cheap one from Tesco does not give a picture and spits errors.

eboston, 2015-02-14:
Just tried the 2-13 nightly on my RPi2 with the raspi-cam.  The video is delayed about 20 seconds.

nophead, 2015-02-14:
raspi-cam works fine for me. The lag is a fraction of a second when viewed over my local network on a PC. It gets to be a lot more over the internet with a phone.

eboston, 2015-02-14:
@nophead This was over a direct wired Ethernet connection on my home network to my PC.  Did you do a test using a Raspberry Pi Model 2?

nophead, 2015-02-14:
Yes 2-13 nightly on RPI model 2 and both Logitech USB and raspi-cam have no significant delay.
I am viewing with Firefox on an XP32 machine.
mpeg_streamer uses about 3% and haproxy 8%.

nophead, 2015-02-14:
```
Feb 14 20:24:55 octopi-red MJPG-streamer [2129]: starting application
Feb 14 20:24:55 octopi-red MJPG-streamer [2129]: MJPG Streamer Version: svn rev: Unversioned directory
Feb 14 20:24:55 octopi-red MJPG-streamer [2129]: fps.............: 10
Feb 14 20:24:55 octopi-red MJPG-streamer [2129]: resolution........: 640 x 480
Feb 14 20:24:55 octopi-red MJPG-streamer [2129]: camera parameters..............:#012
Feb 14 20:24:55 octopi-red MJPG-streamer [2129]: www-folder-path...: ./www/
Feb 14 20:24:55 octopi-red MJPG-streamer [2129]: HTTP TCP port.....: 8080
Feb 14 20:24:55 octopi-red MJPG-streamer [2129]: username:password.: disabled
Feb 14 20:24:55 octopi-red MJPG-streamer [2129]: commands..........: enabled
Feb 14 20:24:55 octopi-red MJPG-streamer [2129]: starting input plugin input_raspicam.so
Feb 14 20:24:55 octopi-red MJPG-streamer [2129]: starting output plugin: output_http.so (ID: 00)
Feb 14 20:24:55 octopi-red MJPG-streamer [2129]: Starting Camera
Feb 14 20:24:58 octopi-red MJPG-streamer [2129]: serving client: 127.0.0.1
```

guysoft, 2015-02-15:
Fixed and released on OctoPi 0.11.0

closed 2015-02-15 by guysoft