guysoft/OctoPi #68: Nightly Build Zipfile Size?

c-born, 2015-02-12:
I tried downloading the latest nightly build, Chrome download conked out at 249MB, (Invalid Zip) so I used wget -c, after a few goes it completed at 520MB, but still Invalid Zip.

I then took a look at the zipfile sizes, and found 49MB, 508MB, 1.0GB, 1.1GB.
I'm assuming 1.1GB is the correct size, and the others are failed builds?
I'll hopefully find out in another 3-4  hours when my download completes, but it would be helpful if either failed builds were deleted, or a note was made somewhere (a read-me on the download page?) as to the approximate size expected for a good Zip file.

c-born, 2015-02-12:
Ok, successfully downloaded 2015-01-31-octopi-0.11.0-2015-02-10.zip, size 1.1GB, still comes up as an invalid archive. Checksums follow, if someone can verify?

  File: 2015-01-31-octopi-0.11.0-2015-02-10.zip
CRC-32: 9b584989
   MD4: 6fc8165d6dd500207c749f9a94c2b313
   MD5: b3d5382415569d26fe0dd743a136631e
 SHA-1: 18519f5308f02fcc0f6de694f8aaae34cb54b0fa

guysoft, 2015-02-12:
I need to add md5 sums so this won't happen. The upload takes 9 hours on the machine and it does not always work

c-born, 2015-02-12:
Understood.
You would generate the sums on your local machine and on the host after upload, then delete the file if the upload was faulty?
Just having the sums available with the files for users to download allows us to check authenticity and correctness, but only after we have finished downloading a possibly corrupt archive.

seefood, 2015-02-13:
also, if you can use rsync in a smart way, you only need to upload a quick delta and not the full file...

seefood, 2015-02-13:
to elaborate - if you use gzip instead of winzip, it has a --rsyncable mode, and it works wonderfully.

guysoft, 2015-02-13:
Added, should work in the next nightly

seefood, 2015-02-15:
well then, we need to change that :-)

guysoft, 2015-02-17:
Md5 Done
@seefood If you want to talk about it then message me, we have plenty of ways to make the infrastructure better.

closed 2015-02-17 by guysoft