guysoft/OctoPi #70: Egalax touchscreen support

tiagojanz, 2015-02-12:
# I usually do this part by ssh  
sudo apt-get update
sudo apt-get upgrade
wget https://github.com/raspberrypi/linux/archive/rpi-3.6.y.tar.gz
tar -zxvf rpi-3.6.y.tar.gz
 sudo apt-get install git libncurses5 libncurses5-dev qt4-dev-tools build-essential
cd linux-rpi-3.6.y/
make mrproper
cp arch/arm/configs/bcmrpi_defconfig .config

# this part need a  x session, so login and startx  than  open terminal   and type
cd linux-rpi-3.6.y/
make xconfig

# look for input devices then select egalax touch   AND touchscreen usb
# click in the floppy icon to save and close the application

sudo make modules_install
sudo cp arch/arm/boot/Image /boot/new_kernel.img

sudo nano /boot/config.txt

######## INSERT #####

kernel=new_kernel.img

####################

######## UNCOMENT #####

#disable_overscan=1

############################

###################### CALIBRATE TOUCHSCREEN  ###########################

sudo apt-get install libx11-dev libxext-dev libxi-dev x11proto-input-dev

 wget http://github.com/downloads/tias/xinput_calibrator/xinput_calibrator-0.7.5.tar.gz

tar -zxvf xinput_calibrator-0.7.5.tar.gz

cd xinput_calibrator-0.7.5

./configure

 make

sudo make install

#then run the application and open the calibration utility in preferences menu (I know you can't test it, but for the sake of completeness) 

#after the calibration finishes it will open a terminal window with the following information 

Section "InputClass"
    Identifier    "calibration"
    MatchProduct    "eGalax Inc. USB TouchController"
    Option    "Calibration"    "42 1959 1837 115"
EndSection

#copy it into 99-calibration.conf

sudo mkdir /etc/X11/xorg.conf.d
sudo nano /etc/X11/xorg.conf.d/99-calibration.conf

# that's it .... reboot  and go

######################################################################

seefood, 2015-02-12:
no need for make xconfig, there was menuconfig last time I checked :)

tiagojanz, 2015-02-12:
didn't know that!!! it could saved me some time.

seefood, 2015-02-12:
Well, when you compile Linux kernel for 21 years, you learn a trick or two :-)
(I actually only ever used make xconfig once and hated it)

seefood, 2015-02-12:
Oh wow... it even has a Wikipedia entry...
https://en.wikipedia.org/wiki/Menuconfig

tiagojanz, 2015-02-12:
hehe did that too ;)

guysoft, 2019-07-02:
Legacy hardware, closing.

closed 2019-07-02 by guysoft