guysoft/OctoPi #72: enabling PiTFT disables boot on Raspberry Pi 2

korneelb, 2015-02-20:
Hi;

I'm using the latest nightly build (the one from the 18th, the 19th and 20th build seem to be corrupted) and everything works on my Raspberry Pi 2.
i expand the filesystem, update the firmware, apt-get update and apt-get upgrade.
so far so good.
as soon as i run the enable-adafruit script in the ~/scripts folder my pi hangs on a reboot on the full color "test" screen..

guysoft, 2015-02-20:
Can anyone confirm this?
@jonaslorander perhaps?

nophead, 2015-02-20:
I ran enable-rpi-display on an RPi 2 and it no longer boots. I think the reason is it installs a new kernel that has not been updated for RPi 2.

korneelb, 2015-02-22:
seems to be a fix here;
https://forums.adafruit.com/viewtopic.php?f=50&t=68854
will try for myself to do this.

PS: is there a tutorial to add the auto update script and the cura slicing engine?

guysoft, 2015-02-22:
@korneelb @nophead Can any of you guys try this fix? I have no hardware (No Pi2 PiTFT )or  to test it on here.
Please write the exact commands you type so we can make a script out of it.
Note: This fix looks really similar to the [Rpi-Display install script](https://github.com/guysoft/OctoPi/blob/devel/src/filesystem/home/scripts/enable-rpi-display), but with different repos

Answer to PS: for Cura there is: https://github.com/foosel/OctoPrint/wiki/Cura-Integration
Please open another issue if you have more questions about that (I didn't understand what you mean by auto-update).

korneelb, 2015-02-22:
yeah i just tried the fix.. while the pitft now works, the octopi script doesn't understand it.. 
so no go :/

guysoft, 2015-02-22:
What do you mean by "script does not understand it". You mean OctoPiPanel does not start?

korneelb, 2015-02-22:
my apologies for the unclarity;

that is correct, the OctoPiPanel script now no longer recognizes the PiTFT so it does not start.

jonaslorander, 2015-02-23:
@korneelb As I do not have a RPi 2 I can not test this, nor do I plan to make an effort without the correct hardware, so I'm counting on the community (you) for this :)
My guess is that the framebuffer has a different name then the one in the OctoPiPanel.py file. Or the driver has changed. 

https://github.com/jonaslorander/OctoPiPanel/blob/devel/OctoPiPanel.py#L98-99

korneelb, 2015-02-24:
yeah i think it's easiest just to wait for the new kernel driver from adafruit. they are working on it, so i'm sure it'll work out..

bolsoncerrado, 2015-02-26:
So to confirm, we can safely burn the latest nightly image and get octopi working on a Raspi2 if we dont have a PiTFT? 

any1 can confirm the HDMI output works with octopi+raspi2?

thanks!

korneelb, 2015-02-26:
I can confirm both

bolsoncerrado, 2015-02-26:
Thanks, just swapped SD onto RPI2 and it does not "work"... it FLIES! :D :D :D

Awesomeee ;)

Now just need a propper touchpanel for 800x480 resolution and we're done hehe

seefood, 2015-02-26:
Small Q: is OctoPi not happy enough with a good old RPiB? will my printer feel any different with an RPI2? because I don't see any reason why the older Pi has any reasons to have bottlenecks, the only bottleneck is really the speed in which the goop can be printed... or am I missing something?

bolsoncerrado, 2015-02-26:
PI2 is a quadcore... PiB is quite capable of handling prints yes, but PI2 gives more processing power for nicer response to a, for example, touch screen with webcam streaming at the same time a timelapse is being recorded. This and the fact everything goes "smoother" in general terms.

But a PIB+ is enough, yes.

korneelb, 2015-02-28:
i know adafruit is working on a fix. i;m getting a new PI2 and will test some more.

korneelb, 2015-03-17:
this is the new library:
https://blog.adafruit.com/2015/03/09/new-pitft-package-beta-release-now-with-pi-2-support-please-try-it-out/

guysoft, 2015-04-11:
Hey guys, any news about this?
Has the issue been solved?

jonaslorander, 2015-04-14:
Hi Guy,

I don't know anything more about this :/

jarrah31, 2016-02-14:
I have exactly the same issue as the original poster, in that when I enable-adafruit script in the ~/scripts folder my pi hangs on a reboot on the full color "test" screen as well. This was with using 2015-05-05-octopi-wheezy-0.12.0.img and the 3.5" PiTFT (Adafruit ID 2097). 

However I have managed to at least boot the Pi with console view by following the instructions found here: https://learn.adafruit.com/adafruit-pitft-3-dot-5-touch-screen-for-raspberry-pi/easy-install , with the following steps after writing the OctoPi image from fresh:

1. curl -SLs https://apt.adafruit.com/add-pin | sudo bash
2. sudo apt-get install raspberrypi-bootloader
3. sudo apt-get install adafruit-pitft-helper
4. sudo adafruit-pitft-helper -t 35r
5. sudo reboot

I'm now going to see if I can get OctoPiPanel working, but at least I'm past the original issue.

Oh and just to confirm, my camera feed is still working at the moment.

guysoft, 2019-07-02:
OctoPiPanel is legacy, closing.

closed 2019-07-02 by guysoft