guysoft/OctoPi #74: Octoprint git pull doesn't work

nophead, 2015-02-26:
Freshly installed OctoPi 2015-01-31-octopi-0.11.0-2015-02-13.zip
```
cd OctoPrint/
git pull
error: Failed connect to gnet.homelinux.com:80; Connection timed out while accessing http://gnet.homelinux.com/git/OctoPrint.git/info/refs
fatal: HTTP request failed

git remote -v
origin  http://gnet.homelinux.com/git/OctoPrint.git (fetch)
origin  http://gnet.homelinux.com/git/OctoPrint.git (push)
```
Shouldn't the remote point to https://github.com/foosel/OctoPrint.git ?

korneelb, 2015-02-26:
i have the same issue with the latest daily builds..

nophead, 2015-02-26:
This fixed it for me
```
git remote rm origin
git remote add origin https://github.com/foosel/OctoPrint.git
git pull origin master
```

korneelb, 2015-02-28:
this did not fix it for me.. i can't pull the latest builds. still getting some errors.

nophead, 2015-02-28:
What sort of errors?

Note the latest OctoPrint is the devel branch, master is the stable one.

korneelb, 2015-02-28:
so the actuall correct fix is
cd ~/OctoPrint
git config --get remote.origin.url  (this will show you the gnet.homelinux.com url) 
git config --replace-all remote.origin.url https://github.com/foosel/OctoPrint.git

run the --get again to verify your change.

nophead, 2015-02-28:
Strange, my fix gives the same end result, so I don't know why it didn't work for you:
```
git config --get remote.origin.url
https://github.com/foosel/OctoPrint.git
```

guysoft, 2015-03-02:
Ok its caused by no cding in the the folder in the command ``git remote set-url``:
Getting in the build: ``Not a git repository (or any of the parent directories): .git``

guysoft, 2015-04-04:
Should be fixed next nightly

guysoft, 2015-04-09:
Still not there, added another fix

guysoft, 2015-04-11:
Fix in last nightly

closed 2015-04-11 by guysoft

foosel, 2015-06-26:
Just for the record, one simple line should actually be enough to fix the URL:

    git remote set-url origin https://github.com/foosel/OctoPrint.git

See also [here](https://help.github.com/articles/changing-a-remote-s-url/)