guysoft/OctoPi #75: Camera not detected one the new raspberry pi 2

vashdansky, 2015-03-01:
Camera not detected one the new raspberry pi 2. Camera works fine on the B+ and B versions. Not detected.. I run a video check and it says supported 1 detected 0???

korneelb, 2015-03-02:
it works for me.. using the latest daily builds..
my problem with the PI2 is (and this seems to be a known issue) that expanding the filesystem on the RPI2 leads to SD Card corruption..

seefood, 2015-03-02:
Is it possible it's a fake SD card? I'd check it out. I have had cards with a reported capacity of 16-128GB turning out to be 7 GB fakes.
http://oss.digirati.com.br/f3/

korneelb, 2015-03-02:
I Wish.. it's a real card, works flawlessly on a Raspberry Pi B or B+ . also a low level format shows it to be 16 gigs...

korneelb, 2015-03-02:
i also ran the test tool and it tells me my card is legit..

seefood, 2015-03-02:
ok then, time to look at the errors that come from the FS resize that fails. does it always happen? does it happen on other cards? does it pop up when doing a full ("slow") format? you want to make sure it's a software thing and not the hardware. Either way, it's a linux distro thing, nothing to do with the OctoPi components, I suppose.

guysoft, 2015-03-20:
I am closing this if the problem is solved in the nightlies - anyone objects?

vashdansky, 2015-03-21:
no.. I must have a bad cable.. the cam did not work on my old one either..

closed 2015-03-21 by guysoft

vashdansky, 2016-01-19:
Hi guysoft/OctoPi,

I'd like to add you to my professional network on LinkedIn.

- Daniel

Confirm that you know Daniel: https://www.linkedin.com/e/v2?e=0-ijlw4bdn-rx&t=ssuw&tracking=eml-guest-invite-cta&ek=invite_guest&sharedKey=wdK_fw0i&invitationId=6095320031246245889

You received an invitation to connect. LinkedIn will use your email address to make suggestions to our members in features like People You May Know. Unsubscribe here: https://www.linkedin.com/e/v2?e=0-ijlw4bdn-rx&t=lun&midToken=AQFmrFVhGEd36Q&ek=invite_guest&loid=AQGBqjHLU0HG_QAAAVJbtWw2DRA1BUPFWnR0pk29xiXVk688gqChxGL450AvM2o14bHtJ3uE3b6wjTpbRIOky8MYpDyfV2qOOiEhOZyrez5nFynaXiSvvLvfA40n9odVrEmsUqtq2r0M9Kf9lWmMB0Kj6ITDrCsnxvI6P-6BnNSm8z-alhRjGQRtVTsUUnrzs3eL&eid=0-ijlw4bdn-rx

This email was sent to reply@reply.github.com.

If you need assistance or have questions, please contact LinkedIn Customer Service: https://www.linkedin.com/e/v2?e=0-ijlw4bdn-rx&a=customerServiceUrl&ek=invite_guest

&copy; 2016 LinkedIn Corporation, 2029 Stierlin Court, Mountain View CA 94043. LinkedIn and the LinkedIn logo are registered trademarks of LinkedIn.

guysoft, 2016-01-19:
@vashdansky Dude you should check your mail, I think you just spammed anyone you ever spoke to,

vashdansky, 2016-01-27:
Daniel Schinsky would like to connect on LinkedIn. How would you like to respond?

Accept: https://www.linkedin.com/e/v2?e=-57oem9-ijvwx28o-56&a=preRegInvite&ek=first_guest_reminder_01&li=14&m=hero&ts=accept_text&sharedKey=wdK_fw0i&invitationID=6095320031246245889

View Daniel Schinsky&#39;s profile: https://www.linkedin.com/e/v2?e=-57oem9-ijvwx28o-56&a=preRegInvite&ek=first_guest_reminder_01&li=3&m=hero&ts=profile_text&sharedKey=wdK_fw0i&invitationID=6095320031246245889

You received an invitation to connect. LinkedIn will use your email address to make suggestions to our members in features like People You May Know. Unsubscribe here: https://www.linkedin.com/e/v2?e=-57oem9-ijvwx28o-56&t=lun&midToken=AQFmrFVhGEd36Q&ek=first_guest_reminder_01&li=16&m=unsub&ts=HTML&eid=-57oem9-ijvwx28o-56&loid=AQGY_aq4st4uFQAAAVKBBhjZGu4uEQAGgKh_DuAnP-P4-DMf5NMBjlRws27u7_UrP7nq1J_I-aoXnMBGFs7pZ2UEXJa9BdOEkX4JtiA4Xy8t5G-T6Hvi3gcqHnmyASAvVm4k9Pjsaw0-1maGSP9InfI6QDzlkY50h3yZDmpKL1M1hsP5imKM9YksEzBM1AawB9AI

This email was sent to reply@reply.github.com.

If you need assistance or have questions, please contact LinkedIn Customer Service: https://www.linkedin.com/e/v2?e=-57oem9-ijvwx28o-56&a=customerServiceUrl&ek=first_guest_reminder_01

&copy; 2016 LinkedIn Corporation, 2029 Stierlin Court, Mountain View CA 94043. LinkedIn and the LinkedIn logo are registered trademarks of LinkedIn.

foosel, 2016-01-27:
Locked since @vashdansky has his LinkedIn profile spamming this thread and the issue is closed anyways.