guysoft/OctoPi #76: Temperature offsets don't do anything

ComputerSmiths, 2015-03-08:
I can change the (bed) offset, but the bed temperature doesn't change, and when I reload the browser the offset reverts to zero. When I set the temperature to 115C with an offset of 15C, the terminal shows the command:
Send: M140 S115.000000
Recv: ok

Am I doing something wrong? Version 1.1.1 (master branch). Same results on 1.1.1-30 (master branch)

Thanks!

guysoft, 2015-03-08:
Sounds like an OctoPrint bug. @foosel ?

ComputerSmiths, 2015-03-08:
I tried to file a bug report on the OctoPrint bug tracker, but the GitBugBot discarded it immediately, and the various FAQ, Guideline, and Bug instructions seem to lead me back here.  I've updated to Version: 1.1.1-32-gd974ab0 (master branch) and it does the same thing, even with a new browser.  Can anyone confirm that it's real?  I can go back through the process of trying to get the GitBugBot to allow me to file a bug report, but I want to make sure it's really a bug...

Thanks!  Love OctoPi!

foosel, 2015-03-09:
In fact, GitIssueBot also told you to please provide anything asked in the template for the bug report ;)

Anyways, there was a typo in the handling of the bed offsets leading to this problem, it's now fixed on the `master` branch (starting with 1.1.1-33).

guysoft, 2015-03-09:
I'd go though the GitBugBot, you are describing a bug.
Just make sure to include logs and things so it would be reproducible.

foosel, 2015-03-09:
@guysoft as I said, it's fixed now, so no need anymore for @ComputerSmiths to go through the bot, and in fact you might want to close this ticket now (or wait until it's included in the OctoPi build, your choice of course)

guysoft, 2015-03-09:
Strange, your replies didn't load here. Closed

closed 2015-03-09 by guysoft