guysoft/OctoPi #77: OctoPi 0.12 nightly appears to be unmodified wheeze

cscott, 2015-03-12:
There should probably be a README file in the http://docstech.net/OctoPiMirror/nightly/ folder explaining the naming scheme.

I downloaded what appeared to be the latest nightly: 2015-02-16-octopi-wheezy-0.12.0-2015-03-11.zip
I installed it on my Raspberry Pi 2 and it seemed to work fine, but there was no octoprint installed.  Poking around the filesystem, I couldn't find any evidence that the octopi build scripts had run, it looks like a completely stock raspian wheeze image.
Perhaps something in the nightly build process broke.

I'm trying what appears to be the latest 0.11.x release now: 2015-01-31-octopi-0.11.0-2015-03-02.zip
Hopefully I'll have better luck with that.

(My understanding is that I need to use a nightly to get Raspberry Pi 2 support, maybe you should do another 'official' release so that's not the case.)

guysoft, 2015-03-12:
Found why,
The last commits made the build fail, and it was not reported, fixing that.

Here is the console output if it interests anyone:
http://gnet.homelinux.com:8080/job/OctoPi/76/console

cscott, 2015-03-12:
Yup, that looks like it would do it.

cscott, 2015-03-12:
Looks like the http://docstech.net/OctoPiMirror/nightly/2015-02-16-octopi-wheezy-0.12.0-2015-03-04.zip build might be the last correct one before the fail?

cscott, 2015-03-12:
hm, no 2015-02-16-octopi-wheezy-0.12.0-2015-03-04.zip doesn't work either  -- the /home/pi/oprint directories are there, but something else is broken. :(

guysoft, 2015-03-13:
Should be fixed next build

closed 2015-03-13 by guysoft