guysoft/OctoPi #78: Upload file to SD does not work

rolo2130, 2015-03-15:
I am running a SeeMeCNC Rostock MAX V2 with Repetier-firmware. I can view the SD card contents. However, I cannot upload or delete files. On upload to the SD card, I receive a Server Internal Error stating that it must be a .GCO or .GCODE file in order to upload. Currently, I'm running the latest nightly build (3/14/2015) and have tried the current release and previous (3/13/2015) nightly. The octoprint.log messages are below.

2015-03-15 06:00:42,010 - octoprint - ERROR - Exception on /api/files/sdcard [POST]
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.9-py2.7.egg/flask/app.py", line 1687, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.9-py2.7.egg/flask/app.py", line 1360, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.9-py2.7.egg/flask/app.py", line 1358, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.9-py2.7.egg/flask/app.py", line 1344, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.2.0_dev_640_g91b9fb3-py2.7.egg/octoprint/server/util/flask.py", line 178, in decorated_view
    return flask.ext.login.login_required(func)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask_Login-0.2.2-py2.7.egg/flask_login.py", line 637, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.2.0_dev_640_g91b9fb3-py2.7.egg/octoprint/server/api/files.py", line 213, in uploadGcodeFile
    filename = fileProcessingFinished(added_file, fileManager.path_on_disk(FileDestinations.LOCAL, added_file), target)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.2.0_dev_640_g91b9fb3-py2.7.egg/octoprint/server/api/files.py", line 189, in fileProcessingFinished
    return filename, printer.add_sd_file(filename, absFilename, selectAndOrPrint)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.2.0_dev_640_g91b9fb3-py2.7.egg/octoprint/printer/standard.py", line 513, in add_sd_file
    remoteName = util.get_dos_filename(filename, existing_filenames=existingSdFiles, extension="gco")
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.2.0_dev_640_g91b9fb3-py2.7.egg/octoprint/util/__init__.py", line 259, in get_dos_filename
    return find_collision_free_name(filename, extension, existing_filenames, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.2.0_dev_640_g91b9fb3-py2.7.egg/octoprint/util/__init__.py", line 313, in find_collision_free_name
    extension = make_valid(extension)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.2.0_dev_640_g91b9fb3-py2.7.egg/octoprint/util/__init__.py", line 310, in make_valid
    return re.sub(r"\s+", "_", text.translate({ord(i):None for i in ".\"/\\[]:;=,"})).lower()
TypeError: expected a character buffer object

guysoft, 2015-03-15:
FYI @foosel

rolo2130, 2015-03-27:
Should I o poo en this up on the Octoprint git?

guysoft, 2015-03-28:
You should, make sure to follow the instructions there or your issue would be automatically closed by the bot there

guysoft, 2015-04-27:
Please move to OctoPrint issue tracker, can't be solved here

closed 2015-04-27 by guysoft