guysoft/OctoPi #79: Image server speed improvement, at least for updates.

seefood, 2015-03-15:
The image server (for the nightlies) is very slow. I get ~70kbps speeds from it, and the file is rather large. Possible solutions are:

1. A daily torrent.
2. Differential delivery:
   * Updates come with apt-get to a vanilla Raspbian.
   * Images compressed with "gzip --rsyncable" rather than PKzip, allowing rsync or zsync
   * rsync on the server side if possible, .zsync if not.

If you are unfamilliar with zsync, it's sort-of rsync but with a smart client working with a dumb server (HTTP or FTP that allows partial GETs). @guysoft would e uploading the image.gz and an image.gz.zsync, and the client would be able to rename the image file he downloaded last time and get only the delta, making repeat downloads much faster. 

I'll add this last part and send you  a pull request. Cheers!

cweagans, 2016-05-01:
Is there any reason to not use the [Releases API](https://developer.github.com/v3/repos/releases/#upload-a-release-asset) to handle all of this? I don't see any reason to have a separate image download server if you can let Github host it for you instead. For the nightly releases, you could just delete the previous nightly and then create a new one with a new uploaded asset. I looked around to see if there was a size limit, but I couldn't find one.

markwal, 2016-05-01:
Actually OctoPrint itself already updates itself and if you switch to the devel branch you can have the excitement of taking every new change as they happen and only the packages that change are updated. Not sure why you need a whole image download every night unless you're testing the image creation process.  The image is really for first time install.

closed 2016-05-01 by guysoft