guysoft/OctoPi #81: enabling autostart of OctoPiPanel

korneelb, 2015-03-17:
Hi;

once i now enable the touchscreen, i have to manually go in and select the octopipanel script at every reboot. can you change the script so it adds itself to the autostart scripts once selected? or perhaps something like
"./scriptname -autostart $true" and ",.scriptname -autostart $false" to enable and disable...

jonaslorander, 2015-03-17:
https://github.com/jonaslorander/OctoPiPanel#automatic-start-up

korneelb, 2015-03-17:
I understand but does this work on OctoPiPanel as well?

korneelb, 2015-03-17:
i mean OctoPi.. since things sometimes work different in this..

nophead, 2015-03-17:
Yes it worked for me.

korneelb, 2015-03-17:
so i did exactly that. but does not start automatically. any way to verify what is happening? running the script manually works.

guysoft, 2015-03-20:
@jonaslorander We should really link that in the OctoPi readme page.
Can anyone who is not me do it?

mikestreety, 2015-09-30:
I tired the OctoPiPanel auto start and doesn't work for me! Any ideas?

guysoft, 2015-10-01:
Can you post what you did exacly and what output you got?

mikestreety, 2015-10-01:
I followed this:

https://github.com/jonaslorander/OctoPiPanel#automatic-start-up

No errors, everything seemed to work fine but doesn't work on boot - the screen just shows the terminal.

jchristman75, 2016-01-30:
Here is what I did for OctoPi, because the service trick did not work for my PiTFT

sudo crontab -e

Then add the line, at the bottom.
@reboot /home/pi/scripts/OctoPiPanel

Save the file (CTRL-O, Enter, CTRL-X) then reboot to test.

guysoft, 2019-07-02:
OctoPiPanel is legacy, closing.

closed 2019-07-02 by guysoft