guysoft/OctoPi #82: Can't connect to octopi at all

AKEric, 2015-03-19:
Very similar to this issue:
https://github.com/guysoft/OctoPi/issues/56
Except I can't connect to the pi at it's address, or any other method suggested:
OSX 10.10.1, Chrome, raspi is connected via wifi.
If I access my router, I can see the local ip of the raspi.  Via the raspi itself, I can browse the internet, so I know it has network access.  But from my mac I get a "This webpage is not available" whenever I try to connect to
http://10.0.0.6/ (local ip)
http://octopi.lan/
http://octopi.local/
https://octopi.local/
It just hangs on "resolving host", the gives the above error.
If I check the running processes on the pi, "octoprint" is listed.
Thoughts?

AKEric, 2015-03-20:
Two modem\router power-cycles later, and turning off my devices wifi and turning it back on, it's connecting like it should.
I think I have a hardware glitch in my router, it's been moody lately.

guysoft, 2015-03-20:
Ok, closing then

closed 2015-03-20 by guysoft