guysoft/OctoPi #83: Automatically triggered 3D print from an online shopping cart purchase? [Request]

LigneDesign, 2015-03-19:
Hello,

<i>[Since this feature is network, and maybe API, related, I'm posting it here first. Let me know if I should post it to the Octoprint new feature request instead (or even somewhere else). Thanks!]</i>

This is a feature request that comes form a need of manufacturers that use 3D printing:
It would be very useful to automatically start a 3D print right after someone bought an object over an online shopping cart; like Volusion for example. 
[That person would have bought a toy (for example) over the online shopping cart. That person is purchasing a real consumer product... not a 3D file, not a gcode file, etc.]
I think that it's a missing link.

A bit about the process:
The object's gcode files would be selected from a gcode database and than queued to some specific 3D printers, of the 3D printer farm.

Manufacturers using 3D printing would use such a feature everyday for their small startup business production. Such a automation would provide a lot more production and growth capacity!
A few manufacturers (www.nrml.com, www.ownphones.com, www.protoseyewear.com) are probably using a similar process to make their products. But I didn't find an opensource solution yet.

What do you think?

guysoft, 2015-03-19:
I got several offers from verious companies to do this. You are more than welcome to write this yourself, I am personally busy with my own startup. If you really want I can link you to others that contacted  me and might be able to build this together, but its not sometimes I'd implement for fun.

guysoft, 2015-03-21:
Closing due to inactivity.

closed 2015-03-21 by guysoft