guysoft/OctoPi #84: Daily builds can't run 250000 baud rate

korneelb, 2015-03-24:
tried this on a raspberry pi B, raspberry pi B+ and raspberry Pi 2.

when using the daily builds and trying to use the 250000 baud rate i get the old octoprint bug 222, can't open serial port.
using the latest stable build works. 

tried the latest stable from februari;
default image - works
running rpi-update and reboot - works
running apt-get update and apt-get upgrade and reboot - works
git pull to Version: 1.2.0-dev-668-g3ebd54d (devel branch) - works

so it's something in the OctoPi distribution..

foosel, 2015-03-24:
Encountered this for some of the nightly builds at work as well. The fix was to add the `python-serial` package to the packages being `apt-get remove`d in `chroot_script`.

The reason appears to be that since the virtual env is created including site packages but then gets its own version of pyserial as well, sometimes the site package one is used, sometimes the local one. I was not able to figure out why that was happening, but just getting rid of the site package (which we don't want need anyways) solved it for good.

guysoft, 2015-03-24:
Merged, please try out the next nightly (starts building only tonight).
Should be build number 94. You can follow progress here: http://gnet.homelinux.com:8080/job/OctoPi/

guysoft, 2015-03-30:
Cam someone with a 250000 baud rate confirm this is indeed fixed?

seefood, 2015-03-30:
I could not with the 1.0.2 merge (what we call "our RBCE" I guess. we need to give it a name). I'm giving a try to the my Development+Rigidboard branch. if it seems to work, I'll try to pull it up to 250000 if you want.

but I still don't see the point. at the speed the step motors are able to move the head, even 38400 should be speedy enough, it's really not that important...

guysoft, 2015-03-30:
Its important for people who can't modify the firmware so freely. Not to mention its a regression. Regressions are not good.

@korneelb ?

korneelb, 2015-03-30:
well, certain printers, like the ultimaker 2, only accept 250000 baud,..

guysoft, 2015-03-30:
Oh, didn't know ultimaker 2 is like that. If anyone with an ultimaker 2 can report the nightly works it would be nice to close this issue.

foosel, 2015-03-30:
@seefood before the stepper motors can move at all, you have to get the data to the firmware, which wants to fill some buffers in order to properly plan ahead its moves. If you have something with a lot of short moves (and hence a lot of lines which don't generate a lot of real movement, but still are at least something like `N384 G1 X20.3 Y83.4 F300*483` <- this long, you can quickly run into issues with low baudrates.

seefood, 2015-03-30:
Well, right now, my Development "bleeding edge" branch is not working, I am back to the Marlin 1.0.2 plus Rigid and both are working only at 115200. As for Ultimaker - that's not our scope, the branch in question is only for RigidBoard.

I'll see if I can figure this out once I dive into the code, but that is going to take a few days/weeks....

nallath, 2015-03-31:
Pyserial is rather rubbish when it comes to being updated. The problem with certain baudrates not being recognised is from 2012 (!!!). 
Doesn't octoPi auto detects the baudrate? Cura uses a fairly nasty way of doing it, but it does work. It simply tries all common baud rates, sends some commands and if the response makes sense, it has to be the correct baud rate.

guysoft, 2015-04-23:
According to #89 this is fixed.
You guys didn't pst this us fixed :-/

closed 2015-04-23 by guysoft

seefood, 2015-08-02:
Just FYI, I have not been able to use 250000 in months. I do fine with 230400 though, which is close enough.

guysoft, 2015-08-03:
Huh, you got a rigidbot to work at 230400?

seefood, 2015-08-03:
Yeah... Why  not? It's a more standard rs232 speed than 250000...