guysoft/OctoPi #86: Logitech C170 does not work

simonCor, 2015-03-27:
The Logitech C170 is listed on https://github.com/foosel/OctoPrint/wiki/Webcams-known-to-work (the comment that it does not work is from me) but does not work.

The error message from mjpg_streamer is the following when i try to start it manually:
Error is: Unable to set format: 1196444237 res: 640x480

The version I used is: 2015-01-31-octopi-0.11.0

guysoft, 2015-03-27:
From what I read here you need to add the -y flag to mjpeg-streamer http://www.raspberrypi.org/forums/viewtopic.php?&t=19661 . Pn a smartphone, but you csn probably find how to set that in /boot/octopi.txt

simonCor, 2015-03-27:
I already tried that, -y did not work. I also tried different resolutions 320x240 etc.

guysoft, 2015-03-27:
Where exactly did you add it? Can you explain in detail and paste the exact line?

simonCor, 2015-03-27:
Thanks for your fast reaction. I will do a detailed description this evening when I am at home.

simonCor, 2015-03-27:
I tried the following changes in /boot/octopi.txt, after each change I rebootet the raspberry pi.
`camera_usb_options="-r 640x480 -f 10 -y"`
`camera_usb_options="-r 320x240 -f 10 -y"`

Here is an example output when I call webcamDaemon:
```
./webcamDaemon
Jan 30 2015 18:28:08
Copyright (c) 2012 Broadcom
version d6e004c61a7a749897c482c860d0b2c28196437e (clean) (release)
camera: auto
usb options: -r 320x240 -f 10 -y
raspi options: -fps 10
~/mjpg-streamer ~/scripts
Running ./mjpg_streamer -o output_http.so -w ./www -i input_uvc.so -r 320x240 -f 10 -y
MJPG Streamer Version: svn rev: Unversioned directory
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 320 x 240
 i: Frames Per Second.: 10
 i: Format............: YUYV
 i: JPEG Quality......: 80
 i: TV-Norm...........: DEFAULT
Unable to set format: 1448695129 res: 320x240
 Init v4L2 failed !! exit fatal
 i: init_VideoIn failed
```

guysoft, 2015-03-28:
Some people say that changing mjpg-streamer helps this.
Can you try using the latest nightly of OcoPi? It has a newer version mjpef-streamer.
Beyond that I am not so sure what to do.

simonCor, 2015-03-30:
Hi,

I tested it with 2015-02-16-octopi-wheezy-0.12.0.img - no luck. The latest octopi build was buggy for me - webinterface didn't schow the inital user creation screen.

guysoft, 2015-03-30:
Does your camera work on raspberrypi or any version of Linux at all?
Perhaps @Joske447 who is said to have tested it there knows?

Joske447, 2015-03-30:
I basically just plugged it in on my RPi with octopi and it worked without any special configuration.

It's doing nothing at all with you?

guysoft, 2015-03-30:
1. can both of you paste the output of `lsusb`? That should make sure this is indeed the same camera with the same chipset.
2. What version of OctoPi @Joske447  ?

simonCor, 2015-03-30:
My lsusb says: Bus 001 Device 006: ID 046d:082b Logitech, Inc.

The cam works on the mac - I currently download debian to test it with it.

Joske447, 2015-03-30:
Terminal on RPi with octopi:

pi@joske-print ~ $ lsusb
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp.
Bus 001 Device 004: ID 148f:5370 Ralink Technology, Corp. RT5370 Wireless Adapter
Bus 001 Device 005: ID 046d:082b Logitech, Inc.

Octopi version:
  a.. Version: 1.1.0-7-g62a51bb-dirty (master branch)

From: simonCor 
Sent: Monday, March 30, 2015 7:39 PM
To: guysoft/OctoPi 
Cc: Joske447 
Subject: Re: [OctoPi] Logitech C170 does not work (#86)

My lsusb says: Bus 001 Device 006: ID 046d:082b Logitech, Inc.

The cam works on the mac - I currently download debian to test it with it.

guysoft, 2015-03-30:
Ok we have established you two indeed have the same cameras, ``046d:082b`` match.

@Joske447 I need the version of OctoPi, not OctoPrint. That is shown if you type ``cat /etc/octopi_version``

Joske447, 2015-03-30:
pi@joske-print ~ $ cat /etc/octopi_version
0.8.0

From: Guy Sheffer 
Sent: Monday, March 30, 2015 11:11 PM
To: guysoft/OctoPi 
Cc: Joske447 
Subject: Re: [OctoPi] Logitech C170 does not work (#86)

Ok we have established you two indeed have the same cameras, 046d:082b match.

@Joske447 I need the version of OctoPi, not OctoPrint. That is shown if you type cat /etc/octopi_version

guysoft, 2015-03-30:
OK, thats a really old version. But its something to start with.
@simonCor You can try and see if your camera works there, if it does it could either be that the raspberrypi version changed since then, or that mjpeg streamer changed. OctoPi 0.8.0 if available here:
http://docstech.net/OctoPiMirror/older_versions/2014-01-07-wheezy-octopi-0.8.0.zip

simonCor, 2015-03-31:
I tried but it seems that it does not boot. My guess: 0.8.0 is not compatible with the raspberry 2

guysoft, 2015-03-31:
Yep, thats mainly why we released 0.11.0, to add support to raspberypi 2.
This leaves two options:

1. Take 0.8.0 to a raspberrypi 1 and upgrade the image firmware using ``rpi-update``.
2. Move the version of  ``/home/pi/mjpg-streamer`` folder in the 0.8.0 image to the 0.11.0/nighlty image and see if that helps. If this does not then it could be that the newer reaspberrypi firmware breaks support to your camera, this is something @Joske447 could test if he tries his camera on 0.11.0.

Joske447, 2015-03-31:
I'm downloading the image now.

To be honest, it's been a while since I worked with octopi (customizing webpage mostly) and linux in general.

I hope I can run the 0.11.0 image on the RPi 1 since I don't have a 2 yet ... I know it's a shame :(

I'll try to test it tomorrow (if I got the time for it :S).

Also, I'm sorry for the fact that I sometimes give vague answers.

I'm guessing you're in a different time zone since your replies to the post are around 0:30 AM here.
So if I don't respond for a day I must have fallen asleep :p 

Oh and does the dist-upgrade work with the OctoPi image or does it only update the debian part (can't remember the update urls).

P.S.: Maybe change the "working" on the supported webcam page.

guysoft, 2015-03-31:
Nope, I am close. Its 2AM here, and will be past 3AM today. Working late on my statup papers :)
Thanks a lot for your help! Without it this issue would have probably hang forever never solved.

You can do ``dist-upgrade``. It should upgrade the debian part.
I don't think ``dist-upgrade`` is needed though when it comes to this issue, the kernel is located in the raspberrypi firmware which is ``rpi-upgrade``.

Joske447, 2015-04-02:
Hey,

I got it working on version 0.11.0 on the RPi 1.

Note that I had to change some settings in mjpg streamer in the file: scripts/webcamDaemon.

# init configuration
camera="auto"
#camera_usb_options="-r 640x480 -f 10"
#camera_raspi_options="-fps 10"
camera_usb_options="-r 1024x768 -f 30"
camera_raspi_options="-fps 30"

The default settings are commented.
I found these settings in my original config too.
This means I'll have to excuse myself for saying it worked out of the box since I obviously changed my settings.

Let me know if you got it working with these settings.

simonCor, 2015-04-02:
Adding these settings to /boot/octopi.conf works!

camera_usb_options="-r 1024x760 -f 30"

Thanks @Joske447 and @guysoft. I will add this to the "webcams known to work" page.

closed 2015-04-02 by simonCor

6ill9re9, 2017-12-23:
Hi, I tried adding both

camera_usb_options="-r 1024x768 -f 30"

and

camera_usb_options="-r 1024x760 -f 30"

the later looks wrong btw
None are giving me a 16/9 ratio, which is very annoying... Maybe I didn't understand this thread and the video format is stuck at VGA...