guysoft/OctoPi #87: Enhancement: uvcdynctrl and guvcview to control webcam

braddo99, 2015-03-29:
Hello,

Love OctoPi - just got a new Raspi2 and got the latest image, wonderful. But, ... I forgot how much time I spent trying to figure out how to control my Logitech (and other uvc) webcams. Turns out there are a lot of things you can do to control uvc webcams from the command line like, in particular, setting the focus distance - makes it much sharper to look at a print in progress.

To make this work, you need to install uvcdynctrl and guvc view, which I had forgotten that I had installed on the previous Rpi - took me a while to figure out why it wasn't working and what to do to get it going!

Now I can type "uvcdynctrl -s "Focus" 100" on the command line and watch the camera adjust focus right in the OctoPrint window. Even better would be a focus adjustment directly from OctoPrint, but that's a suggestion for Gina!

Cheers,

Brad

guysoft, 2015-03-29:
Um, just out of curiosity and hoping to solve another issue, is this related to #86 ?

I guess we can ship uvcdynctrl if it solves people trouble, but what we really need is for this to be documented.

braddo99, 2015-03-29:
I don't think the issue is related. My webcam (Logitech QuickCam 9000 Pro) worked fine out of the box and didn't require the -y flag. However, it's not properly focussed on the print. Many Logitech cams have the ability to change parameters from the command line, including Focus, which I find tremendously useful.

uvcdynctrl allows to change parameters, but Focus is not among them unless you also install guvcview.

so, ...
sudo apt-get install uvcdynctrl
sudo apt-get install guvcview
sudo guvcview (this adds the Focus and a few other controls to uvcdynctrl list)
sudo chmod a+rw /dev/media0 (otherwise can't make the adjustments)
plug and unplug the camera then ...
uvcdynctrl -g "Focus" (get focus setting, which by default is "0" meaning infinite)
uvcdynctrl -s "Focus" 100 (set focus to 100, which I guess is a range from 0 to 255?)

guysoft, 2015-03-31:
This would mean shipping uvcdynctrl and guvcview. 
Also I can't do ``sudo chmod a+rw /dev/media0``, it should be a udev rule. I am not sure if its worth doing.

braddo99, 2015-04-05:
I cannot figure out how to restart "webcamDaemon" without rebooting the whole machine - can someone tell how to do it?

guysoft, 2015-04-05:
This should do it:

    kill -9 `ps aux | grep webcamDaemon | grep -v grep | awk '{print $2'}`

braddo99, 2015-04-05:
awesome, thanks I'll try it out.

Muradamin, 2015-04-16:
THNAKYOU VERY MUCH BRADDO99!!!!
I WAST A LOT OF HOURS TRYING FOCUSING MY WEBCAM AND FINALLY I FOUND THIS THREAD.
Now Im trying to set the focus on 150 (uvcdynctrl -s "Focus" 150 #), but when I restart or shutdown Raspberry, this value comes to the default value.
Can someone tell me to do it?

braddo99, 2015-04-16:
Muradamin, I still have that problem too... with the older version of OctoPi on RPi B+ (not blaming on octopi just think there was something different between that and RPi2 etc etc) I could set the focus this way while OP is running and it would remain even after a reboot of the whole system. Now, I can't set the focus while OP is running, and it's pretty hard to set it at all because you have to shut down mjpg streamer to gain access to the camera. But as soon as you start mjpg streamer again it overrides the focus setting and goes back to blurry. For now I am just living with it but it's very annoying!

braddo99, 2015-04-16:
Actually I think the logitech also has *zoom* control, which would be even more cool, but haven't figured out how to access that because I guess it needs particular drivers and such.

Muradamin, 2015-04-16:
ah, yes.. and also you need to unplug and plug the camera...
Im searching for other sollutions, If I fiind anything I will post it.
Thank you very much braddo99

braddo99, 2015-05-04:
I have seen some related discussion of getting pan/tilt and other functions to work in mjpg-streamer here:

http://sourceforge.net/p/mjpg-streamer/discussion/739917/thread/709f0741/?page=1

and the solution seemed to be to make mjpg-streamer using libv4l2 option like this:

make clean && make USE_LIBV4L2=true

Now, raspbian doesn't have libv4l2 but it seems libv4l-dev contains it according to this:
https://packages.debian.org/sid/libv4l-dev

I wonder if previous versions of OctoPi installed mjpg-streamer with any switches which formerly allowed uvcdynctrl to access the camera and change focus?

I wonder if there's merit to attempting to make mjpg-streamer with USE as above, but not sure what livbrary to call out since libv4l2 isn't explicitly installed as a package.

In the discussion thread it seems doing such an install allowed pan/tilt (and focus/zoom I think) to show up in the actual mjpg-streamer web page. That I would like to see!

braddo99, 2015-05-07:
Well I do wish I could figure out why mjpg-streamer locks out the camera and why unplugging it somehow makes it possible to register the focus control, .... but figured out a work around.

after OctoPi is up and running you have to "unbind" the webcam, which you need to first find out what is the device ID. So run lsusb to find it. Then lsusb -t to see the USB tree, make note of where the camera is according to this syntax bus-port,port.port etc mine is for example 1-1.4.2  Then you can run the following:

sudo bash -c "echo 1-1.4.2 > /sys/bus/usb/drivers/usb/unbind"
then
sudo bash -c "echo 1-1.4.2 > /sys/bus/usb/drivers/usb/bind"
the above is equivalent to unplugging and plugging the usb webcam 
sudo guvcview -a (to add controls to the camera)

Once the webcamDaemon runs again (it's on a timer which repeats every few minutes)
go to (octopi IP):8080 to look at the mjpg-streamer page and confirm that the picture is there. Then look for the "Controls" button. You should now see controls for focus! When you adjust it, you should see the webcam focus change.

If you reboot the pi, you'll have to repeat this process. I still haven't figured out how to automate this even after multiple tries putting these commands into the .bashrc and in the webcamDaemon itself.

mhdshameel, 2016-01-12:
Hi Braddo99,
I have installed uvcdynctrl and am able to control camera settings like brightness, saturation, contrast, sharpness and exposure in my Raspberry pi running Rasbian.
I have installed guvcview and executed the other commands. But if I try to set or get the focus settings I get an error like
"Unknown control specified"
Do you have any suggestions?
Thanx.

BBB, 2016-03-01:
@mhdshameel You need to use the exact name of the control. In my case this was `"Focus, Auto"`. You can see a list of controls by calling `uvcdynctrl -d /dev/video0 -c`

xSvenHD, 2016-11-16:
I'm trying to change my webcam focus settings but im stuck running guvcview. Anyone knows whats going on? The webcam is not being used by octoprint. Im getting this:

`pi@raspberrypi:~ $ sudo guvcview
V4L2_CORE: (UVCIOC_CTRL_MAP) Error: No such file or directory
V4L2_CORE: (UVCIOC_CTRL_MAP) Error: No such file or directory
V4L2_CORE: (UVCIOC_CTRL_MAP) Error: No such file or directory
V4L2_CORE: (UVCIOC_CTRL_MAP) Error: No such file or directory
V4L2_CORE: (UVCIOC_CTRL_MAP) Error: No such file or directory
V4L2_CORE: (UVCIOC_CTRL_MAP) Error: No such file or directory
V4L2_CORE: (UVCIOC_CTRL_MAP) Error: No such file or directory
V4L2_CORE: (UVCIOC_CTRL_MAP) Error: No such file or directory
V4L2_CORE: (UVCIOC_CTRL_MAP) Error: No such file or directory
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.front.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM front
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround40.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround40
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround41
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround50
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround51
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround71.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround71
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM iec958
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM spdif
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM spdif
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
[... cut ...]
jack server is not running or cannot be started
error: XDG_RUNTIME_DIR not set in the environment.
RENDER: Couldn't initialize SDL2: No available video device
RENDER: Couldn't init the SDL2 rendering engine
Unable to init server: Could not connect: Connection refused
GUVCVIEW: (GUI) Gtk3 can't open display
V4L2_CORE: Could not grab image (select timeout): No such file or directory
V4L2_CORE: Could not grab image (select timeout): No such file or directory
V4L2_CORE: Could not grab image (select timeout): No such file or directory
`

Graphix2k, 2017-01-22:
Hello,
I'm running the Logitech 920c and getting the same error as xSvenHD when I type "sudo guvcview", not sure what I'm doing wrong, please help...

szafran81, 2018-01-28:
sudo guvcview -a

then:
uvcdynctrl -d /dev/video0 -c
to get option names

and in my case:
uvcdynctrl -s "Focus, Auto" 0
to turn off autofocus
and then:
uvcdynctrl -s "Focus (absolute)" 50
to set manual focus to value 50

kordris, 2018-01-30:
Is there a way to make a start up file for this? Im kinda noob to linux and stuff related to linux. Seems if you restart the pi these are restarted and I have to re enter these commands to reset my C920 web camera again. Would love to make a script or something to run at startup.

szafran81, 2018-01-30:
sudo nano /etc/rc.local

and this is what I have in mine:
/usr/bin/guvcview -a
/bin/chmod +rw /dev/video0
/usr/bin/uvcdynctrl -s "Focus, Auto" 0
/usr/bin/uvcdynctrl -s "Focus (absolute)" 50
/home/odroid/mjpg-streamer/mjpg-streamer-experimental/mjpg_streamer -o "/home/odroid/mjpg-streamer/mjpg-streamer-experimental/output_http.so -w /home/odroid/mjpg-streamer/mjpg-streamer-experimental/www" -i "/home/odroid/mjpg-streamer/mjpg-streamer-experimental/input_uvc.so -d /dev/video0 -r 1920x1080 -f 5" &

You can use it but you have to change the paths to those on your system (I'm using Odroid with Ubuntu and manually installed OP, and not RPi).

kordris, 2018-01-30:
`/home/odroid/mjpg-streamer/mjpg-streamer-experimental/mjpg_streamer -o "/home/odroid/mjpg-streamer/mjpg-streamer-experimental/output_http.so -w /home/odroid/mjpg-streamer/mjpg-streamer-experimental/www" -i "/home/odroid/mjpg-streamer/mjpg-streamer-experimental/input_uvc.so -d /dev/video0 -r 1920x1080 -f 5" & `

What would be default location for that line on the Octiprint or how would I go and about to find this line?

szafran81, 2018-01-31:
change /odroid/ in paths to /pi/

kordris, 2018-02-21:
Is there a way to check to see if this going off when the pi is starting. I don't think its running? I don't think my pathways are correct. I have to go in and manually turn off the autofocus from time to time. I think its not running. So is there a way to check it and see if it ran correctly? Thanks for the help.

szafran81, 2018-02-21:
Ummmm... I don't really know.
Check each command on your command line if it's working
You can find out where each command is on your system with 'which <command>' to see what paths to use

mdemenok, 2018-03-31:
Instead of installing and running `guvcview`, you can just import the right controls in `uvcdynctrl`:
`uvcdynctrl -i /usr/share/uvcdynctrl/data/046d/logitech.xml`

Psyc0Flyer, 2018-06-20:
I have a c615 and would LOVE to adjust settings in OctoPrint. I have my c615 plugged into my pi 3 B and mounted on my PRUSA i3 MK3 but the focus is not on the print. I was interested in the time-lapse capabilities and have installed the plugin OctoLapse and while a great plugin, I can not control any camera specific settings. Please advise what to install or do to make this functionality work.

TIA!

kordris, 2018-06-20:
Psyc0flyer, not sure if the C615 has the autozoom on it like the C920 or my C930. 
One way to tell is SSh into your pi and run these commands. 

uvcdynctrl -s "Focus, Auto" 0
to turn off autofocus
and then:
uvcdynctrl -s "Focus (absolute)" 50
to set manual focus to value 50

If you havnt installed them just installed them by normal linux commands. But I think there already installed by default.

kordris, 2018-06-20:
This doenst give you options in octiprint to control the zoom and any of the other thing. But it allows my pi to run the script to turn off the autozoom which by default was turn on always. Which really sucked because the camera was always trying to zoom in and out on the moving bed and carriage. Also this allows you to set a custom zoom number so you can tweak your zoom number and get it right were you want it.

Psyc0Flyer, 2018-06-20:
I just found that OctoLapse DOES in fact have options for these settings! Trying now.

[image]

Psyc0Flyer, 2018-06-20:
As you see in this pic, I can't seem to get the cam to focus ON the print :/

[image]

kordris, 2018-06-20:
I never really got around to messing with time laspe stuff. But I'm just starting to mess with it now. So might like to see if that does work. Never saw that in my octoprint. It might not work like you think. I know if you command line it from SSH it will zoom in. But once you restart the pi it will lose your setting unless you setup the startup script.

Psyc0Flyer, 2018-06-20:
I do not know what to do to do from SSH, Advise? OctoLapse has to be added through plugins.

wcndave, 2019-02-25:
I have a Logitech 9000 Pro and wanted to improve the focus.  What I found after reading these threads, with some old data in them, is that I can get this to work by doing some of the things above however slightly differently:

```
ssh to pi
sudo apt-get install uvcdynctrl
sudo apt-get install guvcview
```

I then did **without** sudo
`guvcview `

Then I didn't need to do any un/plugging of camera, just the following:

```
uvcdynctrl -g "Focus" # get focus setting, which by default is "0" meaning infinite
uvcdynctrl -s "Focus" 100 # set focus to 100, which I guess is a range from 0 to 255?
```

So then I tried with all types of values from 0 to 255, and found that 255 was focusing at about 3cm, and 0 being infinite.

I also set some brightness, contrast and saturation values, and got a much nicer picture.

There was no need to unplug, restart, switch off the daemon, nothing... it all just worked.

When one time I had a problem changing resolution, whilst it was streaming, and it said the device was in use, I found this useful

```
pi@octopi:/tmp $ fuser /dev/video0
/dev/video0:           437m
pi@octopi:/tmp $ ps aux1 | grep 437
pi         437  0.1  0.3  26884  3492 ?        Sl   14:26   0:00 ./mjpg_streamer -o output_http.so -w ./www-octopi -n -i input_uvc.so -r 1280x1024 -d /dev/video0
pi         750  0.0  0.0   4368   560 pts/0    S+   14:36   0:00 grep --color=auto 437
pi@octopi:/tmp $ kill -9 437
```

However, it doesn't persist over reboot of pi, (rebooting octoprint seems to have no affect, doesn't kill or modify streaming at all)

I tried adding the two commands to rc.local, however it doesn't work.
If I log it, it shows "segmentation fault".
Put in a pause of 20s first, to let other processes all load, however no affect.

Not quite sure where I go from here.

RicoTrevisan, 2020-10-16:
Err... so I'm not using OctoPi, but it seems that you guys know your way around GUVC. I'm hoping you guys could help me out.

I'm trying to run a webcam through a Pi4 and output the camera's content via HDMI. (like this guy is [doing here](https://www.youtube.com/watch?v=PcbjGvJa73o)). Any ideas how I could accomplish that? Is it possible via the command line?

guysoft, 2020-10-18:
@RicoTrevisan I would appropriate though if you keep on topic.

BTW has this been solved? should I close this or should we add some form of documentation anywhere?