guysoft/OctoPi #88: Unknown webcam not working YUV

joydisee, 2015-04-09:
Hey guys !

Thanks for the great work ! So I got a webcam with LEDs around it (exactly like this one http://www.happy-thingz.com/shops/happythingz/4629644-1.jpg) and I would like to try to use it with Octopi.
So I am currently able to *take pictures using fswebcam*, I also pulled the latest version of mJpeg-streamer and installed it. I then tried to use it with Mjpeg-streamer, it did not work at first, wo I thought, maybe my webcam is a YUYV, so I tried to change the webcamDaemon script to make it YUV compatible... Guess what? It still didnt work! I found a shady post saying that the resolution might be a problem, so I dropped it to 320x240, but it still does not work...
This code-paste summarized my situation pretty well:

```
pi@octopi ~/scripts $ ./webcamDaemon
Mar 16 2015 19:34:31
Copyright (c) 2012 Broadcom
version 51ab816b505d1b745130562908d866915c836056 (clean) (release)
camera: auto
usb options: -r 320x240 -f 10 -y
raspi options: -fps 10
~/mjpg-streamer ~/scripts
Running ./mjpg_streamer -o output_http.so -w ./www -i input_uvc.so -r 320x240 -f 10 -y
MJPG Streamer Version: svn rev: Unversioned directory
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 320 x 240
 i: Frames Per Second.: 10
 i: Format............: YUYV
 i: JPEG Quality......: 80
 i: TV-Norm...........: DEFAULT
Unable to set format: 1448695129 res: 320x240
 Init v4L2 failed !! exit fatal
 i: init_VideoIn failed
~/scripts
```

So what else whould I try ? Should I get a new webcam ? Did I do something wrong?

Thanks !
cc @lussylver

guysoft, 2015-04-12:
Unknown is not that helpful.
Can you paste the output of ``lsusb``?

Also, you can try the solution to this same error given in #86

joydisee, 2015-04-12:
lsusb:

Bus 001 Device 010: ID 0ac8:301b Z-Star Microelectronics Corp. ZC0301 Webcam

Currently trying the fix from #86, brb

joydisee, 2015-04-12:
My problem is that there is no /boot/octopi.conf, I only have a config.txt.
Is that the same? Should I modify the config.txt instead of /boot/octopi.conf ?

```
pi@octopi /boot $ ls
bcm2708-rpi-b.dtb       config.txt     kernel7.img       start_cd.elf
bcm2708-rpi-b-plus.dtb  COPYING.linux  kernel.img        start.elf
bcm2709-rpi-2-b.dtb     fixup_cd.dat   LICENCE.broadcom  start_x.elf
bootcode.bin            fixup.dat      LICENSE.oracle
cmdline-pi-tft.txt      fixup_x.dat    octopi.txt
cmdline.txt             issue.txt      overlays
```

guysoft, 2015-04-12:
Yes the correct name is ``/boot/octopi.txt``

joydisee, 2015-04-12:
Thank you for your time ! So i tried to add the line mentionned in #86 to my `octopi.txt`, it didn't work, same error,

I thought that maybe I should change the script `webDaemon` to the same resolution as in octopi.txt, so i changed this line:
```
camera_usb_options="-r 1024x760 -f 30 -y"
```
and i still got the same error:
```
pi@octopi ~/scripts $ ./webcamDaemon
Mar 16 2015 19:34:31
Copyright (c) 2012 Broadcom
version 51ab816b505d1b745130562908d866915c836056 (clean) (release)
camera: auto
usb options: -r 1024x760 -f 30 -y
raspi options: -fps 10
~/mjpg-streamer ~/scripts
Running ./mjpg_streamer -o output_http.so -w ./www -i input_uvc.so -r 1024x760 -f 30 -y
MJPG Streamer Version: svn rev: Unversioned directory
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 1024 x 760
 i: Frames Per Second.: 30
 i: Format............: YUYV
 i: JPEG Quality......: 80
 i: TV-Norm...........: DEFAULT
Unable to set format: 1448695129 res: 1024x760
 Init v4L2 failed !! exit fatal
 i: init_VideoIn failed
~/scripts
```

I found a post saying that upgrading my RPi firmwares could do the trick, trying it.
Is it not going to work out with this webcam ?

joydisee, 2015-04-12:
Updating the RPi firmware still gives the same error.

guysoft, 2015-04-12:
Try setting the resolution to 640x480, but using the settings from the other issue.

joydisee, 2015-04-15:
So I tried doing that:
added the line `camera_usb_options="-r 640x480 -f 30"` at the end of `/boot/octopi.txt`
and changed back the line in `/scripts/webcamDaemon` to `camera_usb_options="-f 30 -y"
`

Something appears in my octopi webpage, but it is not good:
[image]

guysoft, 2015-04-15:
Remove the -y option. Thats what you get when you put it but dont need iy

joydisee, 2015-04-15:
Yeah !!! It's totally working now !
I'll go add this onw to the list of compatible webcams now !

Thank you very much !

closed 2015-04-15 by joydisee