guysoft/OctoPi #89: Auto Baud Rate Detection Fails

eboston, 2015-04-23:
Just updated to the April 22 nightly (2015-04-22_2015-02-16-octopi-wheezy-0.12.0).  When I went to connect to the printer, I left the baud rate set to AUTO but it failed.  The following was in the octoprint.log:

2015-04-22 19:34:37,511 - octoprint.util.comm - ERROR - Something crashed inside the serial connection loop, please report this in OctoPrint's bug tracker:
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.2.0_dev_758_g2ce40c4-py2.7.egg/octoprint/util/comm.py", line 1033, in _monitor
    self._baudrateDetectTestOk += 1
AttributeError: 'MachineCom' object has no attribute '_baudrateDetectTestOk'

When I set the baud rate manually to 250000, it connected fine.  Not sure if this is an OctoPi or OctoPrint issue.

guysoft, 2015-04-23:
Well at least I can close #84

Looks like this is on @foosel side?

foosel, 2015-04-23:
Already fixed in OctoPrint (I hope, my own tests looked good), you can close that too.

closed 2015-04-23 by guysoft