guysoft/OctoPi #91: Camera is not detected after enabling piTFT screen

CowboyWill, 2015-05-05:
Anyone else having this problem?
When I ran the enable-adafruit-pitft script and the pi rebooted, it now says “No camera detected, trying again in 2 minutes”.  I’m running a 2amp supply on a Raspberry Pi B+ with Pi camera and AdaFruit PiTFT. 

I initially loaded up OctoPi and the camera works great, soon as I run the TFT script it no longer detects the camera.   So I tried loading up a standard raspbian image and ran the AdaFruit scripts to load the piTFT both the camera and display work fine with this image.  I also ran the AdaFruit image and both display and camera work fine.    

I read other problems with power supplies, I have a 2amp supply (I know they don’t always work at rated power) so I found another 2.5amp supply same issue.  Plus it does work with the Adafruit image.   

I tried this several different times with master and devel versions, updated firmware, software, etc.  nothing seems to work.  Tried different camera, different B+, (but only own one display).

Any thoughts on why this might be happening?  Thanks in advance.

Will

guysoft, 2015-05-05:
What camera are you using?

CowboyWill, 2015-05-07:
I have a standard Raspberry Pi Camera version 1.3

mtowara, 2015-10-05:
I see the same behaviour (b+, picam), any fix yet?

markwal, 2015-10-05:
Have you checked with AdaFruit boards?  There must be people not using OctoPrint that are running into the same problem with their PiCam and PiTft?

nophead, 2015-10-05:
I have a b+ with PiTft and PiCam and it works with OctoPi and OctoPiPanel.

guysoft, 2015-10-05:
It might still be a power problem, @nophead are you using wifi/a powered hub?

mtowara, 2015-10-05:
I don't even have the Pitft connected and the daemon running anymore, camera still doesn't work (did before running the pipanel setup script).

There of course always is the chance that I fried my image somehow, had quite some unplanned restarts lately ;)

nophead, 2015-10-06:
I use a Wifi dongle and power from a USB charger.

mtowara, 2015-10-07:
Hi, I fixed it now (also switched to pi 2 but suspect it should also work for b+) by installing pitft by hand (according to this https://github.com/adafruit/Adafruit-PiTFT-Helper and enforcing the adafruit packages).

The setupscript for PiPanel seems broken to me, possibly also related to #72

echo "deb http://apt.adafruit.com/raspbian/ wheezy main" >> /etc/apt/sources.list
wget -O - -q https://apt.adafruit.com/apt.adafruit.com.gpg.key | apt-key add -
apt-get update
apt-get install raspberrypi-bootloader=1.20150904-1
apt-get install adafruit-pitft-helper
adafruit-pitft-helper -t 28r
apt-get install libraspberrypi0=1.20150904-1
apt-get install libraspberrypi-bin=1.20150904-1

guysoft, 2019-07-02:
OctoPiPanel is legacy, closing.

closed 2019-07-02 by guysoft