guysoft/OctoPi #92: Scripts for Plugins

eboston, 2015-05-05:
Would it be possible/practical to add scripts to install common plugins to the OctoPi image?  There are a few that I install every time I burn a new OctoPi image.  In particular, Navbar Temp (https://github.com/imrahil/OctoPrint-NavbarTemp) and OctoPrint-PrintHistory (https://github.com/imrahil/OctoPrint-PrintHistory).

MolarAmbiguity, 2015-05-06:
(I'm not the developer) But it seems that @foosel  is working on a plugin manager. That should solve your problem. (https://github.com/OctoPrint/OctoPrint-PluginManager) However, a script to install a plugin wouldn't be that hard to write (just ```pip install PluginAddress?```)

guysoft, 2015-05-06:
I was planing to do that but @foosel replied she is working on the Plugin Manager which would be the way to install plugins, so for now I am waiting for her to get it to work, then we will bundle it and release.

eboston, 2015-05-06:
Best a question for her, but have you hear what the timeline is for the manager?

I know the scripts would be a simple one line shell command.  Just was hoping to have it in the image so I don't have to keep looking up the command.  Yeah.  I'm being lazy.  :)

foosel, 2015-05-06:
I'm about *this* close to make it a bundled plugin actually ;) just need to look into a sudo option for people who installed octoprint outside of a virtualenv... the plugin as available on github should already be usable very well as it is right now for OctoPi, just not bundled yet.

eboston, 2015-05-06:
@foosel Thanks for the update.  Will look at it in the next day or so.  Got some print jobs that I need to get finished before playing around with changes to Octo.

guysoft, 2015-10-05:
There is a plugin manager now.

closed 2015-10-05 by guysoft