guysoft/OctoPi #93: octopi.local stopped working?

stevelucky, 2015-05-08:
I just installed Octopi for the first time yesterday and all aspects of it were working great. Then, seemingly out of nowhere, I couldn't connect to it via octopi.local. I pulled the SD card and did a full reinstall just in case somehow I messed something up but I still couldn't connect. Finally, I tried connecting via my Raspberry Pi IP address and it connected great. So, for some reason, I just can't connect via octopi.local. Any idea what the issue could be?

stevelucky, 2015-05-08:
How do I delete this thread? I finally saw that someone else had this same issue. I don't want to have a duplicate thread.

stevelucky, 2015-05-08:
Though, I guess if you want to leave it up, I can post my solution. I just edited my hosts file to redirect the IP address to octopi.local. Looked like this:

192.168.1.XXX  octopi.local

After adding that, it worked just fine.

guysoft, 2015-06-03:
Sounds like something with your router setup

foosel, 2015-07-05:
Duplicate of #56, closing add already requested by op

closed 2015-07-05 by foosel