guysoft/OctoPi #94: Running 2 Raspberrys in one Network

TheZippyMan, 2015-05-09:
I got a little problem/issue...

I want to run two Raspberrys with OctoPi connected to one network. How is it possible, that I can connect to 
Raspy 1 with octopi1.local
Raspy 2 with octopi2.local
?

Thank you!

guysoft, 2015-05-09:
Run the following commands on OctoPi to change the hostname:
This example changes the hostname from octopi to octopi1
```bash
echo "octopi1" > /etc/hostname
sed -i -e "s@octopi@octopi1@g" /etc/hosts
```

TheZippyMan, 2015-05-09:
Worked! 
Thank you so much! :)

closed 2015-05-09 by TheZippyMan

nsidious9, 2018-10-08:
dumb question but where do I add this command?  I do not see a command prompt or shell access on my octopi.

guysoft, 2018-10-08:
@nsidious9 Please use the community forum at https://discourse.octoprint.org/