guysoft/OctoPi #95: SSL key generated with 0 bytes size, haproxy start failing

gojibogyo, 2015-05-12:
Somehow during initial startup of the octopi installation (both on stable and almost latest nightly) the ssl key gets generated with 0 bytes size which causes haproxy to fail starting up.
--> raspi not accessible without putting in port 5000 and the links to the raspi camera don't work

foosel, 2015-05-12:
@guysoft FYI, OP came onto OctoPrint IRC channel and we debugged this live. Both the key file and the pem file generated by genCert were 0 bytes, haproxy died with

```
[ALERT] 131/090522 (2124) : parsing [/etc/haproxy/haproxy.cfg:24] : 'bind 0.0.0.0:443' : unable to load SSL private key from PEM file '/etc/ssl/snakeoil.pem'.
[ALERT] 131/090522 (2124) : Error(s) found in configuration file : /etc/haproxy/haproxy.cfg
[ALERT] 131/090522 (2124) : Proxy 'public': no SSL certificate specified for bind '0.0.0.0:443' at [/etc/haproxy/haproxy.cfg:24] (use 'crt').
[ALERT] 131/090522 (2124) : Fatal errors found in configuration.
```

Can't blame it for that. It's not the first time I've heard of people who couldn't access their OctoPi on port 80, so this might be more than just an isolated case. Maybe a lack of entropy during first boot or something, causing the key generation to fail?

Since it's next to impossible to figure out what went wrong there on the very first boot of the Pi in retrospect, my suggestion would be to not only test in genCert if the key file exists but also if has more than 0 bytes in size and regenerate if not.

I could provide a PR for that if that would be ok for you too.

guysoft, 2015-05-13:
A PR would be wonderful. We could just add another AND to the if in : https://github.com/guysoft/OctoPi/blob/devel/src/filesystem/home/scripts/genCert

foosel, 2015-06-02:
Took a bit longer than anticipated due to a full blown cold and consecutive tries at catching up with my TODOs, but there you are, a PR.

guysoft, 2015-06-03:
A fix had been merged,  a reboot should now fix the issue. Should be out in the next nightly. 
Should start at midnight UTC and you can watch it here:  http://gnet.homelinux.com:8080/job/OctoPi/166/

guysoft, 2015-06-10:
Ok last build came out, people are welcome to confirm its fixed (we had a delay due to my laptop being stolen).

foosel, 2015-07-05:
I suggest closing this. Since there's no way to test if it is fixed (for that reliable reproduction first needed to be possible) no one will be able to confirm this and the ticket will stay open forever. Better close it and reopen if it happens again although the fix is in place.

guysoft, 2015-07-05:
Yes, fair enough. Will reopen if someone reports again.

closed 2015-07-05 by guysoft