guysoft/OctoPi #96: Log for mjpegstreamer

arekm, 2015-05-14:
Please run webcam script like this:
````
sudo -u pi -s /bin/sh -c "/home/pi/scripts/webcamDaemon > /home/pi/.octoprint/logs/webcam.log 2>&1" &
````
so we will get some log with info on what is happening.

guysoft, 2015-05-14:
I'm busy these weeks (seed round for my startup). If you can pull request it would get faster in to the nightlies.

arekm, 2015-05-14:
Fresh log with each boot, not sure if we want to keep old entries, too (if so then > needs to be changed to >>).

guysoft, 2015-05-14:
Thanks a lot :)
Next nightly should start in 2 hours from now. And will be visible here (broken untill it starts): 
http://gnet.homelinux.com:8080/job/OctoPi/145/
When its done you welcome to try it out.

guysoft, 2015-05-17:
Should be now in nightly, would appreciate if someone can confirm.

foosel, 2015-07-05:
New webcamd now logs to /var/log/webcamd.log and has logrotate enabled.

@guysoft can we close this or do you want to keep it open until release of 0.12?

guysoft, 2015-07-05:
Closed, and updated the release draft with this issue number.

closed 2015-07-05 by guysoft

OutsourcedGuru, 2020-01-01:
For what it's worth, I'm not seeing the `/var/log/webcamd.log` on **1.4.0rc3 running on OctoPi 0.17.0**.