guysoft/OctoPi #99: In OctoPrint, I'm unable to connect tto the printer

tbolt9000, 2015-05-15:
I have tried two different versions of Octopi with the same results..

I get the error:

Machine State: Error: Failed to open serial port. Check permissions?

tbolt9000, 2015-05-15:
Messages in terminal window below:

Changing monitoring state from 'Offline' to 'Opening serial port'
Connecting to: None
Unexpected error while connecting to serial port: None OSError: '[Errno 2] No such file or directory: 'None'' @ comm.py:_openSerial:948
Changing monitoring state from 'Opening serial port' to 'Error: Failed to open seria...'

nophead, 2015-05-15:
You need to select the correct serial port and baud rate. E.g. /dev/ttyUSB0 or /dev/ttyACM0.

tbolt9000, 2015-05-16:
You are right.

 

I had just figured that out.

 

Thanks,

closed 2015-05-16 by guysoft