guysoft/OctoPi #100: add usbip kernel module

Gateworks, 2015-05-30:
Please add usbip kernel module support so that the distro can also be used to connect printers to remote servers.

guysoft, 2015-05-30:
We are currently not compiling our own kernel, so this would require some work to add. Don't really have the time for such an exotic feature. But would gladly merge pull request if anyone is up for the task

guysoft, 2015-06-07:
A temporary solution would be to use the usb-ip app in the Pi Store. it uses a userspase server that seems to work. However its not open source and lets you connect only one device.

Compiling the kernel in the image is tough because there is only ~2GB free on it. The workaround I can think of would  be to mount an image, build the kernel there, and copy the result.
Guide on how to compile the kernel here: 
https://pietrushnic.github.io/blog/2014/08/18/linux-rpi-and-usb-over-ip/

guysoft, 2018-01-04:
Could be done now with CustomPiOS kernel module