guysoft/OctoPi #101: latest nightly build does not update nor does it start octoprint

korneelb, 2015-06-01:
Hi;

latest nightly builds don't seem to start OctoPrint. i can login over SSH but also when i try to run the update scripts as per the update instructions 

 cd ~/OctoPrint 
2. git pull && git checkout devel 
3. ./venv/bin/python setup.py clean 
4. ./venv/bin/python setup.py install 
5. sudo service octoprint restart 

it cannot find .venv 

hope you can help me.

MolarAmbiguity, 2015-06-01:
I'm running the nightly from 2015-05-31. My octoprint started automatically. To update I followed the instruction for [octopi >= 0.9.0] (https://github.com/foosel/OctoPrint/wiki/FAQ#how-can-i-switch-the-branch-of-the-octoprint-installation-on-my-octopi-image). I'm not sure which builds are octopi >= 1.0.0

guysoft, 2015-06-02:
There seems to be a problem in the nightly builds, not sure if its an internet connection or some change:
You can see the output here: http://gnet.homelinux.com:8080/job/OctoPi/162/console
Will have to look in to it.

cc @foosel 

There is no OctoPi 1.0.0.

guysoft, 2015-06-02:
First error is:

    W: Failed to fetch http://mirrordirector.raspbian.org/raspbian/dists/wheezy/main/binary-armhf/Packages  Hash Sum mismatch

guysoft, 2015-06-03:
It was a hickup from the Israeli Raspbian mirror. Fixe. And today's nightly works.

closed 2015-06-03 by guysoft