guysoft/OctoPi #105: No commands are send to printer

wunnik, 2015-06-06:
It looks like no commands are send to the printer.
Temperature readings are OK but the printere does not head up.

Using: 2015-06-04_2015-02-16-octopi-wheezy-0.12.0

Log from terminal:
Connecting to: /dev/ttyACM0
Changing monitoring state from 'Offline' to 'Opening serial port'
Connected to: Serial<id=0x2ad4570, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from 'Opening serial port' to 'Connecting'
Send: N0 M110*35
Recv: start
Send: N0 M110*35
Recv: echo: External Reset
Recv: Marlin 1.0.0
Recv: echo: Last Updated: Mar 14 2015 10:51:04 | Author: (Adrian at your service!)
Recv: Compiled: Mar 14 2015
Recv: echo: Free Memory: 4126  PlannerBufferBytes: 1232
Recv: echo:Hardcoded Default Settings Loaded
Recv: echo:SD card ok
Changing monitoring state from 'Connecting' to 'Operational'
Recv: ok
Send: N1 M20*16
Recv: Begin file list
Recv: start
Recv: echo:Marlin 1.0.0
Recv: echo: Last Updated: Mar 14 2015 10:51:04 | Author: (Adrian at your service!)
Recv: Compiled: Mar 14 2015
Recv: echo: Free Memory: 4126  PlannerBufferBytes: 1232
Recv: echo:Hardcoded Default Settings Loaded
Recv: echo:SD card ok

guysoft, 2015-06-06:
This is an OctoPrint issue, please report it in the OctoPrint issue tracker https://github.com/foosel/OctoPrint/issues .
Also you better provide which printer is this?

nophead, 2015-06-06:
>Send: N1 M20*16
>Recv: Begin file list
>Recv: start

Your firmware started to send the SD file list and then reset. A firmware or hardware bug unless you pressed the reset button.

guysoft, 2015-06-10:
Unless someone opposes, I am cosing this as not-at-bug

wunnik, 2015-06-13:
I solved the issue, removing the SD card made it working again. Closing the issue.

closed 2015-06-13 by wunnik