guysoft/OctoPi #107: WiFi Configuration

eboston, 2015-06-10:
Something happened to my current OctoPi sd card (couldn't mount something because of corruption) so I figure I would try the latest nightly out.  I got it downloaded and installed with no issues.  My normal procedure is then to boot it up with a keyboard/mouse/monitor and run the raspi-config, enabling the GUI boot.  I would reboot and then run the wireless configuration manager.  However, it seems with the latest version of linux you are using for OctoPi, the wireless configuration manager has been removed or there are no links anywhere to get to it.  I found instructions for setting up a wifi connection manually, but for others not as comfortable editing configuration files, it would be nice to get the wifi manager back.

guysoft, 2015-06-11:
We recently upgraded the version of Raspbian. And there seems to be a change there in the notes regarding 'raspberrypi-net-mods'
http://downloads.raspberrypi.org/raspbian/release_notes.txt
Googeling that I get:
https://www.raspberrypi.org/another-raspbian-desktop-user-interface-update/

eboston, 2015-06-11:
I don't know why I didn't notice that icon the first time.  Probably was too used to the other way and didn't even look for it.  Got it setup and running on WiFi again.  Thanks.

closed 2015-06-11 by eboston