guysoft/OctoPi #110: Hashes of downloads

Findus23, 2015-06-11:
Is it possible to add hashes to the download?
My download stopped multiple times and I had to resume it once with `wget -c`, so I want to check if my files are correct before flashing it.

guysoft, 2015-06-11:
I tried to do it in the Jenkins job, but for some reason it would not upload the md5-sum I have there.
You can take a look at:
http://gnet.homelinux.com:8080/job/OctoPi/

I think there is a bug in Jenkins
It says at the end:

    10:15:07 current root dir /nightly
    10:15:07 current root dir /nightly
    10:15:08 transferred 1 files to /
    10:15:08 Transfered 2 files.

But it only transfers the zip and not the md5

guysoft, 2015-07-11:
Not sure why, Jenkins started uploading the md5 files now. So closing.

closed 2015-07-11 by guysoft