guysoft/OctoPi #122: WPA Supplicant error

Grogyan, 2015-07-09:
2015-07-02_2015-05-05-octopi-wheezy-0.12.0
After inputting in the correct SSID and passkey into /etc/wpa_supplicant/wpa_supplicant.conf OctoPi doesn't load WLan adapter.
Upon opening /etc/network/interfaces found wpa_supplicant had been commented out.
Removing comment causes OctoPi to correctly load WLan adapter

foosel, 2015-07-09:
Yes. That is intentional. You are supposed to configure OctoPi via `/boot/octopi-network.txt`, since that is also reachable without having to boot up the Pi first, thus solving the hen-egg problem of getting the Pi on your network without a keyboard and monitor.

See the [README](https://github.com/guysoft/OctoPi#how-to-use-it).

Grogyan, 2015-07-09:
While I agree that it is logical to have a pre-boot wireless Lan setup.
Problem is people who are used to flashing their PI with Raspbian are not immediately aware that they are to edit a file on the SD card before inserting into the PI. Here, in my case, I have reflashed OctoPi about 5-7 times for various reasons over the past year or so, mostly to do with forgetting the ssh password, and it is only with this build of OctoPi where I wasn't immediately able to get WLan up by inputting the wireless Lan details in headed mode, just by updating wpa_supplicant.conf, before I was sure that everything was setup, then go headless.
Not a big deal either not having the wireless configuration utility, when you can do it via command line.
Confusion arises, when editing the wpa_supplicant.conf doesn't load the WLan adapter as expected.  Then realize that it had been commented out in /etc/network/interfaces.

Surely both methods should be available, headed and headless modes?

foosel, 2015-07-09:
Both methods conflict however. The way the default wpa-supplicant setup is done in `/etc/network/interfaces` takes the wlan interface over exclusively, leaving no chance for other methods to work. At least that's what we found in our tests. The new Raspbian release with its dhcpcd thingy makes it even more difficult to get things working out of the box through traditional means.

It might be possible for a script running before the networking is brought up to detect if the sourced `octopi-network.txt` contains an uncommented `iface` line, hinting at the user having set this up, and only in that case commenting out the wpa line -- and vice versa (no `iface` line => comment that back in).

Grogyan, 2015-07-09:
Well, I will leave it in your capable hands I'll close this, hopefully nobody else will fall for this pitfall, do a Google search, and find this raised issue, pointing back to #how-to-use-it.
Had been careful too, keeping an eye out for important changes in OctoPrint that I needed to be aware of, and didn't notice this change, even though it is in it's own github repo, from a headed mode WLAN setup to headless WLAN

closed 2015-07-09 by Grogyan

foosel, 2015-07-09:
Turns out this wasn't so difficult after all when one uses the `mapping` stanzas available in `/etc/network/interfaces`. PR pushed to only use the new config variant if it's actually set up by the user, otherwise the default one will be used. Unless @guysoft objects this will then be part of 0.13 (or 0.12.1?)

Note that this [nevertheless changed in the 2015-05-05 Raspbian image](https://www.raspberrypi.org/another-raspbian-desktop-user-interface-update/).

guysoft, 2015-07-09:
I support having more than one way to configure the network

mrcleanr6, 2015-07-26:
i have to say i can get the older octopi v1.1.1 up and connected to my netowrk in a matter of about 2 min. the new way with the octopi-netowrk text file is terrible. i have been a week trying to get it to connect and have basically given up.i am so frustrated with this now i have to wipe the sd card clean and go back to the old version.

foosel, 2015-07-26:
It might help if you could clarify where you are failing. Also you can always just follow the steps outlined in the first post above to get back to the stock behavior which - to repeat - has changed with the recent release of Raspbian Octopi is based on, so if it doesn't work with stock either, the fault lies with Raspbian.

mrcleanr6, 2015-07-26:
well i run an open network and uncommenting the 3 lines in the open network section of the text file and adding my ssid didnt have any effect. i know someone else who set it up no problem with a secure network. not sure if the open network settings arent right or what. im not exactly a networking pro. one other thing which may cause confusion for people is that i think most will be opening the text file in notepad. the file looks like a big jumbled mess because its in unix.  it has to be opened in wordpad, text wrangler, text edit, etc. what i ended up doing to get this working was delete the network text file totally. copied the interfaces and supplicant file from my pi running 1.1.1 and made it exactly the same in the new octopi version. boot to the desktop, open a terminal and run wpa_gui. the old wifi config utility and set it up with that. that had no problem finding and connecting to the network. now maybe i did some unnecessary steps there but i was so frustrated messing with this thing when it should have been a 2 min job like in the old version.