guysoft/OctoPi #124: OctoPi unreachable after a while

MichMich, 2015-07-10:
If I don't use my OctoPi server for a while, OctoPi becomes unreachable. Not only via octopi.local, but also via IP. This doesn't seem to be the OctoPrint server since the Pi isn't readable at all (ping, ssh aren't working either.)

How can I figure out what's happening?

Budman17r, 2015-07-10:
I had this issue when the pi was getting low on power.  I had to get a
larger power supply in my case a 5v 1.8 amp wall wart.  Also the USB hub
was interfering with the connection to the printer causing it to crash so
with the larger power supply I plugged it directly in to the pi.

MichMich, 2015-07-11:
I managed to solve it with this tip: 

https://learn.adafruit.com/adafruits-raspberry-pi-lesson-3-network-setup/test-and-configure#fixing-wifi-dropout-issues

guysoft, 2015-07-11:
Hardware spesific issue. Closing and pasting solution here:
Create and edit a new file in /etc/modprobe.d/8192cu.conf

    sudo nano /etc/modprobe.d/8192cu.conf

and paste the following in

    # Disable power saving
    options 8192cu rtw_power_mgnt=0 rtw_enusbss=1 rtw_ips_mode=1
Then reboot with sudo reboot

closed 2015-07-11 by guysoft