guysoft/OctoPi #125: Display Raspberry Pi IP Address on printer display

TheSkorm, 2015-07-16:
Logging this ticket in case OctoPi team has some idea about the below and I don't want to duplicate work. Also logging in case anyone else is looking how to do this.

Ideally I want to hack up Marlin to have an entire octoprint menu display that gets it's values from octoprint via octoprint push gcode but for the time being I have hacked a way to display a message on LCD to display octoprints IP address.

If your network has proper working zeroconf you shouldn't really need this as you can just look up the hostname but if you are using a device that doesn't support it (looking at you Android) then this is helpful.

Put the following in /etc/dhcp/dhclient-exit-hooks.d/octoprint and make executable. When DHCP client runs it'll push the IP address to the display.
```
#!/bin/bash
sleep 10
curl -H "Content-Type: application/json" -H "X-Api-Key: YOURKEYGOESHERE" -X POST -d "{\"command\": \"M117 IP `/bin/hostname -I`\"}" http://localhost:5000/api/printer/command
```

jminardi, 2015-07-26:
Instead of manually entering your key into this command you could extract it from `~/.octoprint/config.yaml`

foosel, 2015-07-26:
E.g. like it is done here: https://github.com/guysoft/OctoPi/blob/devel/src/filesystem/home/pi/scripts/OctoPiPanel#L4

guysoft, 2015-10-13:
This could be a pretty neat plugin

foosel, 2017-02-28:
There is [this plugin](https://github.com/jneilliii/OctoPrint-ipOnConnect), and [this one](https://github.com/dattas/OctoPrint-DetailedProgress) apparently wants to add this functionality as well (or already has).

See also https://github.com/foosel/OctoPrint/issues/1496

closed 2017-02-28 by foosel