guysoft/OctoPi #126: Bananapi SD .img files not working

alejandrosela, 2015-07-21:
I've been trying o install Octopi on a bananapi BP-A20 with no success. I tryied using all the Nightly builds and the final release with no success (2015-05-05-octopi-wheezy-0.12.0-2015-07-02.zip). The images generate a boot partition, and another partition, but the bananapi is not booting at all. I tried installation with mac using the dd command, win32diskimage and RMPrepUSB with the same result. Checksum has been verified, so the file is not corrupted. I tried with two different SD cards (4GB and 8GB). I am running out of options, so I think that it could be a problem with the file itself. 
I´ve been using the bananapi with lubuntu image successfully, so both the SD and the bananapi are working well.

Any other idea? Thanks a lot for the great job sir!

alejandrosela, 2015-08-04:
I managed to install 2015-01-31-octopi-bpi-r1-0.12.0-2015-06-15, it is the only version both from the repository and the nightly builds that has more than 1.1Gb (as reported on #133). Could you please update the official mirror for bananapi with a non corrupted file, or use there the version I reported before?
Thank you so much! After this I think that this issue can be closed.

guysoft, 2015-08-04:
Ok,
I moved them out of the nightly folder and to the official list.
I didn't remove the 07-02 version because I want to test it myself (which might take a while due to me being busy).
But I hope this way at least you have a usable system.

alejandrosela, 2015-08-04:
Thank you very much sir, really appreciated!

guysoft, 2015-08-04:
Thanks for testing :)

guysoft, 2015-10-05:
Fixed

closed 2015-10-05 by guysoft