guysoft/OctoPi #127: update information missing

AndKe, 2015-07-25:
Installed: 1.2.0-dev-939-g8e5c383
Available: 1.2.4 (stable) 

there is a box saying that only "v" marked updates can be installed "directly"
please provide some information about the updates that cannot.

foosel, 2015-07-25:
https://github.com/foosel/OctoPrint/wiki/FAQ#octoprint-tells-me-i-have-an-update-available-but-i-cant-apply-it

closed 2015-07-25 by foosel

AndKe, 2015-07-25:
Thank you for responding.
Yes' I've read that prior to reporting.
This image is the "daily bulid" (I picked from the wrong end of http://docstech.net/OctoPiMirror/nightly/)

-The wiki page you linked to, leads me to https://github.com/foosel/OctoPrint/wiki/Plugin:-Software-Update#making-octoprint-updateable-on-existing-installations

Where I verified setup and config.yaml contents.
- Anyway . I'll proceed with getting proper nightly build from the end of the list, as this is most likely not a problem anymore.

foosel, 2015-07-25:
June 12th indeed might have been too old to already come properly configured.

Just be aware that nightlies can be broken - they do reflect development snapshots after all.

AndKe, 2015-07-25:
yes, I missed current stable, somehow nighly was easier to find :) - works fine now, based on build 2015-07-02   while nightly  2015-07-24 did not respond on port 80   / no OctoPrint were accessible, ssh ok.