guysoft/OctoPi #128: Feature request: enable GPIO pin control for hardware control.

AndKe, 2015-07-25:
OctoPi works fine. 
I am tempted to let it be always-on in my storage facility, however, having a bigger PSU and RepRap board, always on is a waste of power, and enabled high power/heat components to be powered on in case of some bizarre  fault.

I would suggest integrating a GPIO pin interface in the software, that could easily, using few components, control the main power supply for the printer.

foosel, 2015-07-25:
OctoPi is the SD card image for a Raspberry Pi that among other things includes OctoPrint. What you are asking for sounds more like OctoPrint. Note that what you want to achieve can already be done easily through OctoPrint's [System Commands](https://github.com/foosel/OctoPrint/wiki/Controlling-a-relay-board-from-your-RPi) and that there's also always an option to write a plugin if that doesn't suit the desired usecase. But an OctoPi issue this is not, and [there also already exists an OctoPrint issue for that](https://github.com/foosel/OctoPrint/issues/273).

closed 2015-07-25 by foosel

AndKe, 2015-07-25:
thank you. 
I had the impression that OctoPi was the place to ask because it was Raspberry Pi  specific.

foosel, 2015-07-25:
GPIOs aren't necessarily Raspberry Pi specific :) Also, think of it this way - as soon as it doesn't (only) involve bundling additional software with the SD card but also adding functionality to the web interface, it stops being an isolated OctoPi thing and OctoPrint is the more fitting tracker.