guysoft/OctoPi #131: OctoPiPanel. is it only supported with adafruit pitft

Bigfellas, 2015-07-30:
Can you tell me does you build allow the us of any other touch screen. Have borrowed a Waveshare spot pear V3 and was wondering if it will work. Or should I just go ahead and buy a Adafruit.
Thanks
Matt

guysoft, 2015-08-02:
It also works with rapi-display by watterott.
Where are the drives for your "Waveshare spot pear V3" located I can't find them when I search.

MolarAmbiguity, 2015-08-02:
@guysoft there are drivers for the Waveshare Spot Pear V3 here: https://github.com/swkim01/waveshare-dtoverlays

Bigfellas, 2015-08-02:
I have ordered a Pitft, So don't go to too much trouble. I have been playing around with a friends Waveshare and there is not a lot of support out there. So I spent the extra.

Thanks
Matt

MolarAmbiguity, 2015-08-02:
Yeah, the manufacturer (waveshare) doesn't support their product very well. Your are probably better off with someone else.

guysoft, 2015-08-04:
Closing since this seems to be nothing we will not do in this case (unless someone with access to the hardware is up for it).

closed 2015-08-04 by guysoft