guysoft/OctoPi #133: Nightly Builds are Broken

mikenew12, 2015-08-02:
Just installed the build from 31-Jul-2015 onto a new RP2. As far as I can tell, Octoprint isn't actually installed anywhere. ```~/.octoprint``` is present, but no Octoprint. The download size was also 1.06 GB rather than 1.13 GB for the stable build, which looking back I guess should have been suspicious.

I assume the build is failing and uploading an incomplete image. Maybe failed builds could be flagged, or just not uploaded at all?

alejandrosela, 2015-08-03:
Could be related to my issue with the bananapi images. I see that there are a lot nightly builds but none of them worked.

guysoft, 2015-08-03:
Added a fix to jenkins, it might help. It seems like internet connection problems on update are causing this.

seefood, 2015-08-03:
now how about those zsync files?

guysoft, 2015-08-03:
Its an option, but I think we might just rsync, the file size is not the cause, its the connectivity. I seems to me like its the mirror fault since I upgraded the connection here.

seefood, 2015-08-03:
that's why zsync. it requires you to only add a line to the jenkins'
script, and change nothing on the webserver.

guysoft, 2015-08-04:
Update from @foosel (which can't type until she gets better):
She started working on a new Jenkins build server which should make everything work far more smoothly. 
I tried fixing the nightly builds here, but hopefully, perhaps as soon as next week, she might be on getting shiny new Jenkins build server working.

mikenew12, 2015-08-04:
@guysoft In the meantime, maybe there could be a note in the ReadMe about the builds being incomplete? It's pretty painful to go through the download/copy/boot/setup only to realize there's parts missing.

Sorry to hear about @foosel, hope she's better soon.

guysoft, 2015-08-05:
I have yet to find a place people read before they download.

foosel, 2015-08-05:
Trust me when I say, there isn't one. people don't read. not even when you force them in from of a neon pink flashing message box. Give up on the idea ;)

(wrist starting to feel better btw, still forcing myself to refrain from typing though.... tough ;))

guysoft, 2015-08-05:
We could add this:
https://wiki.jenkins-ci.org/display/JENKINS/Embeddable+Build+Status+Plugin
People like colors.

mikenew12, 2015-08-05:
I vote 'yes' on the little colored indicators.

seefood, 2015-08-05:
why do you need pink blinking signs if the build failed? it should not upload in that case... this is build systems 101. Guy, do you need help on that?

guysoft, 2015-08-05:
The problem is it fails on the upload. And no one checks the md5.

Guy

mikenew12, 2015-08-05:
@guysoft Would the build status plugin actually help in this case? If it's failing on upload, then a big green "build-passing" indicator might just make it harder for anyone to realize that the builds are incomplete. It took me a while to realize Octoprint wasn't on the image simply because I didn't think to look.

seefood, 2015-08-06:
If the upload fails, why not remove the half-uploaded image on the server?

guysoft, 2015-08-06:
Because it the jenkins job crashing

guysoft, 2015-10-05:
Fixed

closed 2015-10-05 by guysoft