guysoft/OctoPi #134: "Print" button enabled on connection, before file selected.

AndKe, 2015-08-03:
Once the printer is connected, the "print" button is enabled, and it have no function.
This is the status after connection.
Machine State: Operational
File:  
Timelapse: -
Approx. Total Print Time: -
Print Time: -
Print Time Left: -
Printed: -

guysoft, 2015-08-03:
You should file OctoPrint bugs in the OctoPrint issue tracker.
Also, what version of OctoPi?

AndKe, 2015-08-03:
oops , my bad. browser suggested a github url while typing "octop, and my brain was running NOP's only. :)

my OctoPi is 1.2.4  , downloaded recently a nightly that were working (not the recent ones)
the 3D printer is outside my home, so I got quickly to get full control, and can control the lights and master PSU as well (the one that provides 12V ,30Amax).
I must say OctoPi was a pleasure to set up and use, it's almost too simple, so "thank you, great job"

guysoft, 2015-08-04:
OctoPi version is 0.12.
OctoPrint version is 1.2.4.
You should really report this to on OctoPrint issue tracker, closing here.

closed 2015-08-04 by guysoft