guysoft/OctoPi #135: Help setting up pitft capacitive touchscreen

Bigfellas, 2015-08-04:
Can I get help setting up the Capacitive version of the PTtft.
I tried using the script. But it doesn't work I am guessing it is for the resistive version only.

So I run the setup on the Adafruit page 
https://learn.adafruit.com/adafruit-2-8-pitft-capacitive-touch/easy-install
 
I can now get Page to display but I don't appear to have touchscreen. I runt the config and it tells me 

pi@octopi ~ $ ~/scripts/calibrate-rpi-display
xres = 320, yres = 240
selected device is not a touchscreen I understand
Took 1 samples...
Top left : X = -1225509860 Y = -1094484708
Took 1 samples...
Top right : X = -1225509860 Y = -1094484708
Took 1 samples...
Bot right : X = -1225509860 Y = -1094484708
Took 1 samples...
Bot left : X = -1225509860 Y = -1094484708
Took 1 samples...
Center : X = -1225509860 Y = -1094484708
-0.222646 -0.236613 0.264939
0.491203 -0.202054 0.226243
Calibration constants: -14591 -15506 17363 32191 -13241 14827 65536

Can anyone help. 
Thanks
Matt

guysoft, 2015-08-04:
@jonaslorander  ?

jonaslorander, 2015-08-05:
That is the instruction for resistive screen. Instead of running
sudo adafruit-pitft-helper -t 28r
Try
sudo adafruit-pitft-helper -t 28c

As stated in the script help.
https://github.com/adafruit/Adafruit-PiTFT-Helper/blob/master/adafruit-pitft-helper

Also, I would suggest you download the latest weezy and see if you can get the pitft working there in case the resistive pitft including issues on Octopi when trying to use the capacitive.

Good luck and let us know the result!

Bigfellas, 2015-08-06:
If I run startx I can use the touch screen. Just not when Octopipanel screen is on

Bigfellas, 2015-08-06:
Ok so it looks like the Touchscreen is working. I can hit he reboot.

But as soon as it is connected to the printer I get this error.

Please let me know if I should move this over to the OctopiPanel Github

pi@octopi ~ $ ~/scripts/OctoPiPanel
sh: 1: cannot create /sys/class/gpio/gpio252/direction: Directory nonexistent
sh: 1: cannot create /sys/class/gpio/gpio252/value: Directory nonexistent
OctoPiPanel initiated
OctoPiPanel started!
---
Traceback (most recent call last):
  File "/home/pi/OctoPiPanel/OctoPiPanel.py", line 596, in <module>
    opp.Start()
  File "/home/pi/OctoPiPanel/OctoPiPanel.py", line 194, in Start
    self.get_state()
  File "/home/pi/OctoPiPanel/OctoPiPanel.py", line 293, in get_state
    self.HotEndTemp = state['temps']['tool0']['actual']
KeyError: 'temps'
pi@octopi ~ $

Bigfellas, 2015-08-07:
Ok so I did an update via Git just to check I was on the dev and I was but after I did this. I am all operational. No maybe not the latest version was getting installed?
So now running on Rpi b+ with a capacitive touch screen.

jonaslorander, 2015-08-07:
Great! I emails in am answer asking you about the versions and branches but it seems not to have been added to the comment list.

Good that it works!

closed 2015-10-13 by guysoft