guysoft/OctoPi #138: Add samba and share the watched folder

foofoodog, 2015-08-11:
The watched folder should be available right off the bat to admin users.
On first run, enabling access control and adding the admin user, the share should be out there.
Or just leave the smbpasswd at pi/raspberry

guysoft, 2015-10-13:
Hey,
That would be nice, you can already access the timeplase on the web UI or via ssfs.
If anyone wants to implement this they are welcome to send a pull request.

marianoronchi, 2016-01-22:
I make it works this way:

sudo apt-get install samba samba-common-bin
sudo nano /etc/samba/smb.conf
-------------------------------------------------------------------------
[uploads]
comment = OctoPrint Time Lapses
path = /home/pi/.octoprint/uploads
writeable = Yes
only guest = Yes
create mask = 0777
directory mask = 0777
browseable = Yes
public = yes

[watched]
comment = OctoPrint Time Lapses
path = /home/pi/.octoprint/watched
writeable = Yes
only guest = Yes
create mask = 0777
directory mask = 0777
browseable = Yes
public = yes

[timelapse]
comment = OctoPrint Time Lapses
path = /home/pi/.octoprint/timelapse
writeable = Yes
only guest = Yes
create mask = 0777
directory mask = 0777
browseable = Yes
public = yes

[octoprint]
comment = OctoPrint Time Lapses
path = /home/pi/.octoprint
writeable = Yes
only guest = Yes
create mask = 0777
directory mask = 0777
browseable = Yes
public = yes
-------------------------------------------------------------------------
sudo service samba restart
sudo smbpasswd -a pi

After this you can map your folders in windows using //[YourPiHostname]

Source: http://theurbanpenguin.com/wp/index.php/setting-up-a-samba-server-on-raspberry-pi/ and http://simonthepiman.com/how_to_setup_windows_file_server.php

viper1619, 2016-08-09:
2 changes I had to make for my set-up:

1) guest only = no 
2) samba restart syntax changed

`sudo apt-get install samba samba-common-bin
sudo nano /etc/samba/smb.conf
|------------------------------------------------------------------
[uploads]
comment = OctoPrint Time Lapses
path = /home/pi/.octoprint/uploads
writeable = Yes
only guest = No
create mask = 0777
directory mask = 0777
browseable = Yes
public = yes

[watched]
comment = OctoPrint Time Lapses
path = /home/pi/.octoprint/watched
writeable = Yes
only guest = No
create mask = 0777
directory mask = 0777
browseable = Yes
public = yes

[timelapse]
comment = OctoPrint Time Lapses
path = /home/pi/.octoprint/timelapse
writeable = Yes
only guest = No
create mask = 0777
directory mask = 0777
browseable = Yes
public = yes

[octoprint]
comment = OctoPrint Time Lapses
path = /home/pi/.octoprint
writeable = Yes
only guest = No
create mask = 0777
directory mask = 0777
browseable = Yes
public = yes
|------------------------------------------------------------------
sudo service smbd restart
sudo smbpasswd -a pi`

Lepterois, 2017-06-11:
Hi!
It's working perfectly with filezilla, but with windows I can only read the files, I cannot write in the folder...

An idea?

Thanks

Wolfedelic, 2017-10-07:
Hey Lepterois, in case you didn't solve it or in case anyone else like myself comes across this problem

Try adding the lines 

guest ok = yes
read only = no

To the conf for the folders you want to access. Fixed the inability to write from windows share folder for me

mcmasterp, 2017-11-28:
Just a heads up to anyone using this I had to do:

` sudo apt-get update`

before installing samba or it would fail

Lepterois, 2017-12-12:
Hi all!
Thanks for this replies! Sorry I was out the subject for a while!
I try all of this as soon as I 've 5 minutes!
thanks for all!

mcmasterp, 2017-12-25:
so weirdly this stopped working on new octopi setups. i got two new printers and set them up just as I did my other two and I can't write from my windows 10 machine. It tried both versions on conf file contents shown here and adding the lines @Wolfedelic suggested, still no go.

thisiskeithb, 2018-02-18:
Here's the latest block of code that allowed me to access shares on the pi from both Windows and Mac computers.

```
sudo apt-get update && sudo apt-get upgrade
sudo apt-get install samba samba-common-bin
sudo nano /etc/samba/smb.conf
```
|---------insert the following at the bottom of smb.conf----------------------------------------------
```
[uploads]
comment = OctoPrint Time Lapses
path = /home/pi/.octoprint/uploads
writeable = Yes
only guest = No
create mask = 0777
directory mask = 0777
browseable = Yes
public = yes
guest ok = yes
read only = no

[watched]
comment = OctoPrint Time Lapses
path = /home/pi/.octoprint/watched
writeable = Yes
only guest = No
create mask = 0777
directory mask = 0777
browseable = Yes
public = yes
guest ok = yes
read only = no

[timelapse]
comment = OctoPrint Time Lapses
path = /home/pi/.octoprint/timelapse
writeable = Yes
only guest = No
create mask = 0777
directory mask = 0777
browseable = Yes
public = yes
guest ok = yes
read only = no

[octoprint]
comment = OctoPrint Time Lapses
path = /home/pi/.octoprint
writeable = Yes
only guest = No
create mask = 0777
directory mask = 0777
browseable = Yes
public = yes
guest ok = yes
read only = no
```
|--------------end insert into smb.conf ------------------------------------------------
```
sudo service smbd restart
sudo smbpasswd -a pi
```
_(enter a password to access shares)_

guysoft, 2018-02-18:
@foosel Wanna add this to the image? Or at least ship a script?
(sorry to bug on Sunday, expecting an answer after)

foofoodog, 2018-02-18:
@thisiskeithb Maybe use a different comment for each share.  And maybe only watched needs to be writable.

foofoodog, 2018-02-18:
Oops on the close.

thisiskeithb, 2018-02-18:
@foofoodog I've updated my comment to include different comments for each share and made the watched folder the only one writable.

mcmasterp, 2018-02-18:
I like to have the uploads writable as well

thisiskeithb, 2018-02-18:
@mcmasterp @foofoodog This is probably why all the shares were writable to start. Everyone will have a different preference.

foofoodog, 2018-02-19:
@thisiskeithb Except I believe writing to uploads directly is not supported since it skips the cataloging of metadata that happens in the transition from watched to upload.  As for the others, I would have to be convinced that someone who only is only using SMB needs the ability to manage the whole installations file system. And please note the original feature request mentioned only the watched folder so all other folders added are out of scope.

foosel, 2018-02-19:
@guysoft In principle yes but I'd definitely leave out the settings folder (maybe keep it in the config as an option but commented out by default). Considering that who can read the settings folder also has access to the user account data and stuff like the global API key, this makes sharing this way too sensitive of a matter, even if only with a log in. Power users who know what they are doing still might want to, they can add this then, but I don't think that your regular user should ever need to manually edit something in there and I fear what might happen if they can ("I fumbled around with this file I found somewhere and now my OctoPrint won't start anymore, help me!"). Loaded shotgun, meet foot ;)

I actually had planned to take a closer look at what's the current state with regards to wifi and such and maybe send some PRs your way today, I can see if I can give this a test too in the process of this.

mcmasterp, 2018-02-22:
@thisiskeithb @foosel @foofoodog 

I tried keith's code but I still get access denied on my win10 machine. It's very strange because i used the original instructions on a few pi3 with no issues then all of a sudden one I setup would give me access denied when I try to write to it. Since then every pi3 i've setup has had the same issue. I tried slight alterations of the settings with no luck. I was really hoping keith's instructions would work for me. Anyone else having this experience? I tried connecting to the share as guest and as the pi user with password.

foosel, 2018-02-22:
Odd, for some reason Github didn't show me any of the comments between the last from me and @guysoft until now 🤔

In any case, @foofoodog is right, writing from external programs to the uploads folder is currently unsupported and can cause issues, so no, that should not be writable. Metadata is one part of the reason, file name sanitisation another.

Also I haven't gotten around to look at this after all, the other stuff took up too much time, so can't confirm nor deny any of the above examples working on my end.

mcmasterp, 2018-02-22:
@thisiskeithb @foosel 

I got it to work! I had to alter keiths conf file to make uploads writable yes and read only no and now the share can be written to from my win10 file explorer

foosel, 2018-02-22:
As I just explained, writing to the uploads folder from anything but the OctoPrint process is unsupported. Feel free to do it anyhow, but don't open tickets for any issues caused by that ;) Use the watched folder instead.

mcmasterp, 2018-02-22:
@foosel Oh I see now. I misunderstood the point of the watched folder. I thought anything you uploaded there would automatically start printing. My mistake, only because in other unrelated software I've used in the past a "watched" folder is acted upon without user interaction. makes sense now. 

Also when you said "writing from external programs" I thought you meant programs such as slicing software like S3D saving with a post processing script. again my fault.

VibroAxe, 2018-02-26:
Just as an observation, because of how samba works the access issues above can probably be solved with the following
```chmod -R 777 /home/pi/.octoprint/uploads
chmod -R 777 /home/pi/.octoprint/watched```

Samba applies both samba permissions and then linux permissions, so if you are doing a guest access it needs both guest access within samba AND all-write permissions in the file system (the final 7 if the above chmod string). From my initial investigation it doesn't look like the .octoprint/uploads and /watched dirs are created as 777 by default by octoprint!

foosel, 2018-02-26:
> From my initial investigation it doesn't look like the .octoprint/uploads and /watched dirs are created as 777 by default by octoprint!

Intentionally. If you set 777 on something, you should have a *really* good reason for that as you are allowing everyone and their mother on your Linux system to do everything in that folder. 

This sounds rather like having to get the user mapping right on Samba's side (and I wouldn't even allow guest access here, that's only asking for trouble, especially if combined with any kind of automation on uploaded files), not fiddle around with folder permissions.

VibroAxe, 2018-02-26:
@foosel totally agree, it's the right decision. However the original post seemed to imply that they are attempting to get the write using guest mode. Might have been wrong on that front.

rrauenza, 2018-04-02:
Samba has a force group / force user / force create option for a particular share.  I've used this to force fields dropped into a shared folder to be writable by a shared group.

You'll have to decide what security model you want to use, but when you pick it, samba can likely accommodate it without opening the folder to 777.