guysoft/OctoPi #139: Gcode only case-insensitive for 0 argument commands

eovnu87435ds, 2015-08-17:
I'm not sure if this is an issue or not, but I found that if you type a gcode command with no arguments into the terminal page, it works whether the letter is upper or lower-case(g28 and G28 both work). Commands with multiple arguments only work if each letter is uppercase. (M280 P0 S150 works, m280 p0 s150 does not).

I noticed that if I use a different host program, such as Repetier host, upper and lower case both work.

markwal, 2015-08-17:
OctoPrint and OctoPi are related, but distinct.  OctoPrint is the host software.  OctoPi is the image and image making software.

This isn't an issue with the image (OctoPi) or the image making process.  If it is an issue, it is either the host software (OctoPrint) or the firmware.  As you mention, since it works with Repetier host, it may be that Repetier host uppercases input before sending to the firmware, so it's probably worth looking at in OctoPrint.  I suggest you enter an issue there (https://github.com/OctoPrint).  But when you do, please click the "Contributing Guidelines" link at the top of the issue page to copy and paste the bug form.

eovnu87435ds, 2015-08-17:
Thanks, moved issue report to fossel/Octoprint. Closing issue here.

closed 2015-08-17 by eovnu87435ds