guysoft/OctoPi #141: Timelapse Not Working

CMDann, 2015-08-29:
Hey guys, looking for some help. My timelapse videos are not working right now. I have tried changing the FPS multiple times but I am told this doesn't work. I get this error:

```
Rendering failed
Rendering of timelapse acme_20150829022702.mpg failed with return code 1

More 
[mpeg2video @ 0x1fb78c0] MPEG1/2 does not support 10/1 fps
Error while opening encoder for output stream #0:0 - maybe incorrect parameters such as bit_rate, rate, width or height
```

[image]

I am running the latest OctoPi stable release on my Raspberry Pi model B. My printer is a modified Printrbot 1405 but that doesn't really affect anything. Is there a specific FPS that IS supported? I have tried everything. I was able to get a few good videos before it just flat out stopped working.

markwal, 2015-08-29:
The default fps for timelapse is 25.  I don't know what the minimum is without some experimentation but I see a lot of internet chatter complaining about this same error message with 15 (not with OctoPrint, just people using ffmpeg to encode videos).  So I'm guessing 10 is too low.  You were trying to slow it down I guess?  You could achieve the same result by taking a snapshot more often (like every 5 secs instead of 10).

CMDann, 2015-08-29:
Thanks for the help, it seems to be working now:

https://www.youtube.com/embed/qGudpmB9RCw

Setting the rate to 25 works fine. This is not an issue with OctoPrint or OctoPi just FFMPEG.

closed 2015-08-29 by CMDann